inkcut/inkcut #275: Sketch support

DeeEmm, 2020-09-12:
Hi Guys,

Unlike many I do not use inkscape to design. Instead I use a program called Sketch (an OSX only program).

I'd really like to add a plugin to be able to send my design directly from Sketch to Inkcut.

I'm guessing that opening a file in inkcut via the CLI is  simply a case of invoking inkcut via the CLI with the relevant filename / path.

What I'd like to know is what commands are available to the CLI. Calling inkcut with the -h, -help switches does not return any usable info.

TIA

/DM

frmdstryr, 2020-09-12:
`inkcut open <path>` or use - for the path to read from stdin

DeeEmm, 2020-09-12:
Thanks @frmdstryr I will try to get my head around the sketch plugin format and have a bit of a play. I'll report back here.

Franck78, 2020-09-25:
@frmdstryr , 

add a "-d name" option to select the device to use.