inkcut/inkcut #280: Side pane tabs (Materials, Weedlines) not opening on Linux

ayushchand, 2020-09-29:
Issue seems to be with enaml crashing on linux with Qt 5.15.0. Downgraded to Qt 5.14 and issue is resolved.

https://github.com/nucleic/enaml/commit/0d08198c4493df57a123540e4b8b88cedcc905be

Franck78, 2020-10-02:
they open but :

* with the mouse, it is a very small area you have to click in
* click on the panel; nothin happens; spacebar it opens

I don't have any crashes

```
2020-10-02 18:10:09,642 | INFO    | inkcut | Version: 2.1.2dev
2020-10-02 18:10:09,642 | INFO    | inkcut | Python: 3.6.10 (default, Jan 16 2020, 09:12:04) [GCC]
2020-10-02 18:10:09,647 | INFO    | inkcut | System: uname_result(system='Linux', node='opensuse', release='5.3.18-lp152.41-default', version='#1 SMP Thu Sep 3 23:02:59 UTC 2020 (a4d139b)', machine='x86_64', processor='x86_64')
2020-10-02 18:10:09,647 | INFO    | inkcut | Executable: /usr/bin/python3
2020-10-02 18:10:09,647 | INFO    | inkcut | Args: ['/usr/bin/inkcut']
2020-10-02 18:10:09,676 | INFO    | inkcut | Qt: 5.15.1 Api:

```

ayushchand, 2020-10-03:
I have verified it again, and the issue for me on Qt5.15.1 seems to be this

- click on the panel; nothing happens; spacebar it opens

This is not present in Qt5.14 and clicking works