inkcut/inkcut #282: Output settings in default_config for new devices are not loading

ayushchand, 2020-10-03:
device/drivers/manifest.enaml file lists a default_config for Silhouette devices with Output settings.

```
        DeviceDriver:
            manufacturer = 'Silhouette'
            model = 'Cameo 4'
            width = '300cm'
            length = '30.5cm'
            protocols = ['gpgl']
            connections = ['printer']
            default_config = {
              'scale': [ 5.64, 5.64 ],
              'swap_xy': True,
              'mirror_x': True,
              'mirror_y': True
            }
```

Creating a new device from GUI does not initialize the output settings.

Schubisu, 2021-03-28:
Thanks @ayushchand,
I've just received my Cameo 4 and faceed this issue. As a workaround I deleted the `default_config` from `device/drivers/manifest.enaml` and added it manually in the GUI again.

karliss, 2022-09-26:
I just noticed this as well when working on different issue. After adding some debug logs it seemed like part of problem was that code responsible from initializing stuff based on default config gets executed only when creating a new device before you have selected driver. So your new device gets initialized with default config from whatever driver happens to be first in the list. 

This needs to be solved with care since user might want to keep some of the settings when switching driver or at least write them down. Maybe some kind of confirmation window when you do it: "Do you want to apply default settings from the selected driver, previous settings will be lost? Yes/No/Cancel".

PR #359 (2022-11-12): Rework device creation to use all the data from driver.
* add UI for choosing driver before device is created
* disable driver switching after device is created
* add command for copying a device

In theory the driver switching after device is created can't work for general case where the specific driver 'DeviceDriver(Declarative)' has custom factory which creates some `Device(Model)` subclass like it is for PiDevice. In practice it doesn't properly work even in the simple cases as indicated by #282 .  Switching driver after device has been created also introduces problem of how to deal with u [... cut]
closed 2022-11-12 by frmdstryr