inkcut/inkcut #294: Crashes on kunbuntu 20.10

cactusbloem, 2020-12-08:
After installing inkcut without errors, I get the following error error when I start inkcut:

attr connection 
    patch_pyserial_if_needed()
  File "/usr/local/lib/python3.8/dist-packages/inkcut/device/transports/serialport/plugin.py", line 40, in patch_pyserial_if_needed
    dummy = ListPortInfo()
TypeError: __init__() missing 1 required positional argument: 'device' 

Before Kubuntu I installed MX-Linux with KDE Plasma 5, there i have the same problem.

My machine information:

i7 Core, 8 Gb RAM, nvidia Quadro

Kubuntu 20.10,  plasma 5.19.5 ,  KDE-framework 5.74.0,  Qt-version 5.14.2, Kernel-version 5.8.0.31-generic

Has anyone an solution for this?

frmdstryr, 2020-12-08:
The patch needs removed or you can downgrade pyserial to 3.4

frmdstryr, 2020-12-08:
I pushed an update so you can `pip3 install -U inkcut` and this hack is removed.

cactusbloem, 2020-12-08:
Thanks for the quick response.

I  did run the update but I have got no result, I got this after I  run the update:

self.set_text(d.text)
  File "/usr/local/lib/python3.8/dist-packages/enaml/core/declarative_meta.py", line 43, in __call__
    value = engine.read(owner, name)
  File "/usr/local/lib/python3.8/dist-packages/enaml/core/expression_engine.py", line 179, in read
    return pair.reader(owner, name)
  File "/usr/local/lib/python3.8/dist-packages/enaml/core/standard_handlers.py", line 101, in __call__
    return call_func(func, (tr,), {}, scope)
  File "/usr/local/lib/python3.8/dist-packages/inkcut/joystick/view.enaml", line 72, in 
    device.position[0],
AttributeError: 'NoneType' object has no attribute 'position'

frmdstryr, 2020-12-08:
Can you try removing any of the json files in `~/.config/inkcut`? Looks like the configuration may have got messed up...

cactusbloem, 2020-12-11:
I tried everything nothing seems to work.

I removed everything and restored my system with timeshift. I followed the install  guide step by step but I have no luck getting this app working on Linux. 

I have an question for the experts:

installing this:   apt-get install python3-pip python3-pyqt5 python3-setuptools libcups2-dev python3-pyqt5.qtsvg
goed without error, everything is ok.

then I try to install this:

pip3 install inkcut

in the beginning everything is oke but at the end I got this errors:

Failed building wheel for pycups

Failed building wheel for enaml

Failed building wheel for enaml

compilation terminated.
    error: command 'x86_64-linux-gnu-gcc' failed with exit status 1
    
    ----------------------------------------
Command "/usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-8gm4wlqx/atom/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-record-pyj8zhgy/install-record.txt --single-version-externally-managed --compile --user --prefix=" failed with error code 1 in /tmp/pip-install-8gm4wlqx/atom/

I do not understand why, because the first installation was to meet the depencies, that installation went without errors, so what is the problem?

frmdstryr, 2020-12-11:
Can you attach the full output?

cactusbloem, 2020-12-11:
[error_inkcut.txt](https://github.com/inkcut/inkcut/files/5680856/error_inkcut.txt)

frmdstryr, 2020-12-11:
Thanks. Can you install the python3-dev package with apt and retry?

cactusbloem, 2020-12-12:
> Thanks. Can you install the python3-dev package with apt and retry?

Thanks, that was the solution to get rid of the installation errors, so yes the installation now is without errors.

But when  I excute inkcut in terminal, it seems to to start but after a few seconds  Python crashes.  inkcut give this error:

mo@Fujitsu-MX-Lin:~
$ inkcut
Logging to /home/mo/.config/inkcut/logs/inkcut.txt
2020-12-12 09:31:25,466 | INFO    | inkcut | ========================================
2020-12-12 09:31:25,466 | INFO    | inkcut | Inkcut launched
2020-12-12 09:31:25,466 | INFO    | inkcut | Version: 2.1.3
2020-12-12 09:31:25,466 | INFO    | inkcut | Python: 3.7.3 (default, Jul 25 2020, 13:03:44) 
[GCC 8.3.0]
2020-12-12 09:31:25,471 | INFO    | inkcut | System: uname_result(system='Linux', node='Fujitsu-MX-Lin', release='5.6.0-2-amd64', version='#1 SMP Debian 5.6.14-2~mx19+1 (2020-06-14)', machine='x86_64', processor='')
2020-12-12 09:31:25,471 | INFO    | inkcut | Executable: /usr/bin/python3
2020-12-12 09:31:25,471 | INFO    | inkcut | Args: ['/home/mo/.local/bin/inkcut']
2020-12-12 09:31:25,498 | INFO    | inkcut | Qt: 5.11.3 Api:
2020-12-12 09:31:25,498 | INFO    | inkcut | ========================================
QSettings::value: Empty key passed
QSettings::value: Empty key passed
KCrash: crashing... crashRecursionCounter = 2
KCrash: Application Name = python3.7 path = /usr/bin pid = 22905
KCrash: Arguments: 
KCrash: Attempting to start /usr/lib/x86_64-linux-gnu/libexec/drkonqi from kdeinit
sock_file=/run/user/1000/kdeinit5__0

[1]+  Gepauzeerd              inkcut
mo@Fujitsu-MX-Lin:~

The Python crash information you can find it here: 

[python3.7-20201212-093126.kcrash.txt](https://github.com/inkcut/inkcut/files/5682730/python3.7-20201212-093126.kcrash.txt)