inkcut/inkcut #295: Having trouble installing on Windows...

gtoal, 2020-12-09:
I believe I've followed the instructions accurately.  Finding the Visual Studio tools was tricky but that seemed to be OK too.  However I still get errors installing.  

The final error message was: 

`ERROR: Command errored out with exit status 1: 'c:\users\user\appdata\local\programs\python\python39\python.exe' -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\User\\AppData\\Local\\Temp\\pip-install-iv36izjd\\twisted_7f54b21069be477dad60c2db78b03c7a\\setup.py'"'"'; __file__='"'"'C:\\Users\\User\\AppData\\Local\\Temp\\pip-install-iv36izjd\\twisted_7f54b21069be477dad60c2db78b03c7a\\setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record 'C:\Users\User\AppData\Local\Temp\pip-record-ggbljbiu\install-record.txt' --single-version-externally-managed --compile --install-headers 'c:\users\user\appdata\local\programs\python\python39\Include\twisted' Check the logs for full command output.`

Full console output attached.  It's two consecutive attempts, with an installation of Visual Studio tools between the first and second attempts. (separated by a line starting with '============' ...)

[inkcut install problems.txt](https://github.com/inkcut/inkcut/files/5668826/inkcut.install.problems.txt)

I couldn't find the logs mentioned ( C:\Users\User\AppData\Local\Temp\pip-record-ggbljbiu\install-record.txt did not exist ) - I've also attached a listing of C:\Users\User\AppData\Local\Temp\ in case there are any helpful files in there that you may need to analyze this problem.
[missing error log.txt](https://github.com/inkcut/inkcut/files/5668871/missing.error.log.txt)

I'm an experienced programmer (C & unix primarily) but not a python coder or a Microsoft coder, but for practical reasons I need to drive my vinyl cutter from a Windows portable.  So installation errors may be naïve newbie errors rather than something obscure.

Thank you in advance if anyone is willing to have a quick look at this and see if there's an obvious problem with an easy fix!

Graham

frmdstryr, 2020-12-10:
It looks like twisted doesn't support python 3.9 yet. You can try to uninstall it and reinstall 3.8.

gtoal, 2020-12-10:
Thanks, that did the trick.  There were a lot of older versions to choose
from so I picked 3.8.1 from about a year ago to be on the safe side.

Probably should update the installation page to point away from the default
python installation until the incompatibilities are found.

Now to see if I can drive any of my equipment with it! (KNK Groov-e, Cricut
or 3018 CNC...)

Graham