inkcut/inkcut #296: Not all options available in 'connection' menu

raboof, 2020-12-12:
On 2.1.3, Under Device->Setup->Connection, the 'Type' pull-down does not always seem to be populated the same way: when selecting the Roland CX-24 driver on the 'General' tab, it sometimes has all 6 options, but sometimes only 3 (I didn't record which ones, but missed the 'Raw device descriptor' option in particular).

Shutting down inkcut and starting it again made all options appear again...

frmdstryr, 2020-12-18:
Thoughts on just removing the filtering altogether?

raboof, 2020-12-19:
I'm not opposed to it, but it looks like we use the same concept for the protocols, and there it seems quite useful.

(I understand this is not a very actionable issue in its current form - I'm also OK with just closing it until we find more information on what triggers it)