inkcut/inkcut #297: RuntimeError:  [from .plugin import PreviewPlugin]

TracyB-3091, 2020-12-13:
I'm just starting up Inkcut for the first time and got the following error.  Please help me know how I can fix this:

from .plugin import PreviewPlugin
  File "c:\users\mom and dad\appdata\local\programs\python\python39\lib\site-packages\inkcut\preview\plugin.py", line 12, in <module>
    import pyqtgraph as pg
  File "c:\users\mom and dad\appdata\local\programs\python\python39\lib\site-packages\pyqtgraph\__init__.py", line 19, in <module>
    import numpy  ## pyqtgraph requires numpy
  File "c:\users\mom and dad\appdata\local\programs\python\python39\lib\site-packages\numpy\__init__.py", line 305, in <module>
    _win_os_check()
  File "c:\users\mom and dad\appdata\local\programs\python\python39\lib\site-packages\numpy\__init__.py", line 302, in _win_os_check
    raise RuntimeError(msg.format(__file__)) from None
RuntimeError: The current Numpy installation ('c:\\users\\mom and dad\\appdata\\local\\programs\\python\\python39\\lib\\site-packages\\numpy\\__init__.py') fails to pass a sanity check due to a bug in the windows runtime. 
[inkcut.txt](https://github.com/inkcut/inkcut/files/5684003/inkcut.txt)

frmdstryr, 2020-12-13:
Please see https://github.com/inkcut/inkcut/issues/295 . Some of the dependencies don't work on windows with python 3.9.

TracyB-3091, 2020-12-13:
I uninstalled Python 3.9 and installed Python 3.8.1.  Everything worked without error except the Inkcut when I run it from CMD either as administrator or my user.  What am I doing wrong?

[image]
[inkcut.txt](https://github.com/inkcut/inkcut/files/5685470/inkcut.txt)