inkcut/inkcut #298: Not working with atom.api 0.6.0

parkerlreed, 2020-12-16:
Arch Linux
python-atom 0.6.0-3

```
[parker@wolfcola inkcut-2.1.3]$ inkcut
Logging to /home/parker/.config/inkcut/logs/inkcut.txt
2020-12-16 04:03:47,877 | INFO    | inkcut | ========================================
2020-12-16 04:03:47,877 | INFO    | inkcut | Inkcut launched
2020-12-16 04:03:47,877 | INFO    | inkcut | Version: 2.1.3
2020-12-16 04:03:47,877 | INFO    | inkcut | Python: 3.9.1 (default, Dec 13 2020, 11:55:53) 
[GCC 10.2.0]
2020-12-16 04:03:47,877 | INFO    | inkcut | System: uname_result(system='Linux', node='wolfcola', release='5.9.14-zen1-1-zen', version='#1 ZEN SMP PREEMPT Sat, 12 Dec 2020 14:36:44 +0000', machine='x86_64')
2020-12-16 04:03:47,877 | INFO    | inkcut | Executable: /usr/bin/python
2020-12-16 04:03:47,877 | INFO    | inkcut | Args: ['/usr/bin/inkcut']
2020-12-16 04:03:47,888 | INFO    | inkcut | Qt: 5.15.2 Api:
2020-12-16 04:03:47,888 | INFO    | inkcut | ========================================
2020-12-16 04:03:47,944 | ERROR   | inkcut | Traceback (most recent call last):
  File "/usr/lib/python3.9/site-packages/inkcut-2.1.3-py3.9.egg/inkcut/app.py", line 97, in main
    from inkcut.core.workbench import InkcutWorkbench
  File "/usr/lib/python3.9/site-packages/inkcut-2.1.3-py3.9.egg/inkcut/core/workbench.py", line 22, in <module>
    from enaml.workbench.ui.api import UIWorkbench
  File "/usr/lib/python3.9/site-packages/enaml/workbench/ui/api.py", line 8, in <module>
    from .action_item import ActionItem
  File "/usr/lib/python3.9/site-packages/enaml/workbench/ui/action_item.py", line 8, in <module>
    from atom.api import Bool, Dict, Typed, Unicode
ImportError: cannot import name 'Unicode' from 'atom.api' (/usr/lib/python3.9/site-packages/atom/api.py)

Traceback (most recent call last):
  File "/usr/bin/inkcut", line 33, in <module>
    sys.exit(load_entry_point('inkcut==2.1.3', 'console_scripts', 'inkcut')())
  File "/usr/lib/python3.9/site-packages/inkcut-2.1.3-py3.9.egg/inkcut/app.py", line 97, in main
    from inkcut.core.workbench import InkcutWorkbench
  File "/usr/lib/python3.9/site-packages/inkcut-2.1.3-py3.9.egg/inkcut/core/workbench.py", line 22, in <module>
    from enaml.workbench.ui.api import UIWorkbench
  File "/usr/lib/python3.9/site-packages/enaml/workbench/ui/api.py", line 8, in <module>
    from .action_item import ActionItem
  File "/usr/lib/python3.9/site-packages/enaml/workbench/ui/action_item.py", line 8, in <module>
    from atom.api import Bool, Dict, Typed, Unicode
ImportError: cannot import name 'Unicode' from 'atom.api' (/usr/lib/python3.9/site-packages/atom/api.py)
```

parkerlreed, 2020-12-16:
Tried using atom 0.5.0 as mentioned here but it fails to compile https://github.com/inkcut/inkcut/issues/238#issuecomment-609541024

raboof, 2020-12-16:
I think this should be fixed by https://github.com/frmdstryr/enamlx/commit/16df11227b8cee724624541d274e481802ea67e3 - are you on version v0.4.5 of enamlx?

parkerlreed, 2020-12-16:
Oddly it's on 0.4.5

```
Requirement already satisfied: enamlx in /usr/lib/python3.9/site-packages (0.4.5)
Requirement already satisfied: enaml in /usr/lib/python3.9/site-packages (from enamlx) (0.10.2)
Requirement already satisfied: future in /usr/lib/python3.9/site-packages (from enaml->enamlx) (0.18.2)
Requirement already satisfied: kiwisolver>=1.0.0 in /usr/lib/python3.9/site-packages (from enaml->enamlx) (1.3.1)
Requirement already satisfied: qtpy>=1.3 in /usr/lib/python3.9/site-packages (from enaml->enamlx) (1.9.0)
Requirement already satisfied: ply>=3.4 in /usr/lib/python3.9/site-packages (from enaml->enamlx) (3.11)
Requirement already satisfied: atom>=0.4.1 in /usr/lib/python3.9/site-packages (from enaml->enamlx) (0.6.0)
Requirement already satisfied: setuptools in /usr/lib/python3.9/site-packages (from enaml->enamlx) (51.0.0)
```

parkerlreed, 2020-12-16:
Checking enamlx manually the import has the updated Str

```
from atom.api import (
    Int, Enum, Bool, Str, Typed,
    Coerced, Event, Property, ForwardInstance, observe
)
```

parkerlreed, 2020-12-16:
enaml itself may be the issue (rereading my initial error text)

@raboof What enaml version are you on?

raboof, 2020-12-16:
> enaml itself may be the issue (rereading my initial error text)
> 
> @raboof What enaml version are you on?

0.11.2

parkerlreed, 2020-12-16:
Ok that sorted all the enaml errors but now bytecode is killing it, bleh.

bytecode version?

0.11.0 here

```
[parker@wolfcola ~]$ inkcut 
Logging to /home/parker/.config/inkcut/logs/inkcut.txt
2020-12-16 05:30:48,892 | INFO    | inkcut | ========================================
2020-12-16 05:30:48,892 | INFO    | inkcut | Inkcut launched
2020-12-16 05:30:48,893 | INFO    | inkcut | Version: 2.1.3
2020-12-16 05:30:48,893 | INFO    | inkcut | Python: 3.9.1 (default, Dec 13 2020, 11:55:53) 
[GCC 10.2.0]
2020-12-16 05:30:48,893 | INFO    | inkcut | System: uname_result(system='Linux', node='wolfcola', release='5.9.14-zen1-1-zen', version='#1 ZEN SMP PREEMPT Sat, 12 Dec 2020 14:36:44 +0000', machine='x86_64')
2020-12-16 05:30:48,893 | INFO    | inkcut | Executable: /usr/bin/python
2020-12-16 05:30:48,893 | INFO    | inkcut | Args: ['/usr/bin/inkcut']
2020-12-16 05:30:48,904 | INFO    | inkcut | Qt: 5.15.2 Api:
2020-12-16 05:30:48,904 | INFO    | inkcut | ========================================
2020-12-16 05:30:49,010 | ERROR   | inkcut | Traceback (most recent call last):
  File "/usr/lib/python3.9/site-packages/bytecode/instr.py", line 187, in _set
    opcode = _opcode.opmap[name]
KeyError: 'END_FINALLY'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.9/site-packages/inkcut-2.1.3-py3.9.egg/inkcut/app.py", line 99, in main
    workbench.run()
  File "/usr/lib/python3.9/site-packages/inkcut-2.1.3-py3.9.egg/inkcut/core/workbench.py", line 147, in run
    from enaml.workbench.core.core_manifest import CoreManifest
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "/usr/lib/python3.9/site-packages/enaml/core/import_hooks.py", line 151, in exec_module
    code, _ = self.get_code()
  File "/usr/lib/python3.9/site-packages/enaml/core/import_hooks.py", line 412, in get_code
    return self.compile_code()
  File "/usr/lib/python3.9/site-packages/enaml/core/import_hooks.py", line 377, in compile_code
    code = EnamlCompiler.compile(ast, file_info.src_path)
  File "/usr/lib/python3.9/site-packages/enaml/core/enaml_compiler.py", line 179, in compile
    return compiler.visit(node)
  File "/usr/lib/python3.9/site-packages/enaml/core/enaml_ast.py", line 359, in visit
    result = visitor(node, *args, **kwargs)
  File "/usr/lib/python3.9/site-packages/enaml/core/enaml_compiler.py", line 195, in visit_Module
    self.visit(item)
  File "/usr/lib/python3.9/site-packages/enaml/core/enaml_ast.py", line 359, in visit
    result = visitor(node, *args, **kwargs)
  File "/usr/lib/python3.9/site-packages/enaml/core/enaml_compiler.py", line 218, in visit_EnamlDef
    code = EnamlDefCompiler.compile(node, cg.filename)
  File "/usr/lib/python3.9/site-packages/enaml/core/enamldef_compiler.py", line 241, in compile
    return compiler.visit(node)
  File "/usr/lib/python3.9/site-packages/enaml/core/enaml_ast.py", line 359, in visit
    result = visitor(node, *args, **kwargs)
  File "/usr/lib/python3.9/site-packages/enaml/core/enamldef_compiler.py", line 249, in visit_EnamlDef
    first_code, index_map = FirstPassEnamlDefCompiler.compile(
  File "/usr/lib/python3.9/site-packages/enaml/core/enamldef_compiler.py", line 58, in compile
    compiler.visit(node)
  File "/usr/lib/python3.9/site-packages/enaml/core/enaml_ast.py", line 359, in visit
    result = visitor(node, *args, **kwargs)
  File "/usr/lib/python3.9/site-packages/enaml/core/enamldef_compiler.py", line 97, in visit_EnamlDef
    cg.pop_top()                            # helper -> name -> base
  File "/usr/lib/python3.9/contextlib.py", line 124, in __exit__
    next(self.gen)
  File "/usr/lib/python3.9/site-packages/enaml/core/code_generator.py", line 385, in try_squash_raise
    bc.Instr("END_FINALLY"),                     # TOS
  File "/usr/lib/python3.9/site-packages/bytecode/instr.py", line 137, in __init__
    self._set(name, arg, lineno)
  File "/usr/lib/python3.9/site-packages/bytecode/instr.py", line 189, in _set
    raise ValueError("invalid operation name")
[... 32 lines cut ...]
    result = visitor(node, *args, **kwargs)
  File "/usr/lib/python3.9/site-packages/enaml/core/enaml_compiler.py", line 218, in visit_EnamlDef
    code = EnamlDefCompiler.compile(node, cg.filename)
  File "/usr/lib/python3.9/site-packages/enaml/core/enamldef_compiler.py", line 241, in compile
    return compiler.visit(node)
  File "/usr/lib/python3.9/site-packages/enaml/core/enaml_ast.py", line 359, in visit
    result = visitor(node, *args, **kwargs)
  File "/usr/lib/python3.9/site-packages/enaml/core/enamldef_compiler.py", line 249, in visit_EnamlDef
    first_code, index_map = FirstPassEnamlDefCompiler.compile(
  File "/usr/lib/python3.9/site-packages/enaml/core/enamldef_compiler.py", line 58, in compile
    compiler.visit(node)
  File "/usr/lib/python3.9/site-packages/enaml/core/enaml_ast.py", line 359, in visit
    result = visitor(node, *args, **kwargs)
  File "/usr/lib/python3.9/site-packages/enaml/core/enamldef_compiler.py", line 97, in visit_EnamlDef
    cg.pop_top()                            # helper -> name -> base
  File "/usr/lib/python3.9/contextlib.py", line 124, in __exit__
    next(self.gen)
  File "/usr/lib/python3.9/site-packages/enaml/core/code_generator.py", line 385, in try_squash_raise
    bc.Instr("END_FINALLY"),                     # TOS
  File "/usr/lib/python3.9/site-packages/bytecode/instr.py", line 137, in __init__
    self._set(name, arg, lineno)
  File "/usr/lib/python3.9/site-packages/bytecode/instr.py", line 189, in _set
    raise ValueError("invalid operation name")
ValueError: invalid operation name
```

raboof, 2020-12-16:
> bytecode 0.11.0 here

bytecode 0.11.0 here as well. We're building with python 3.8 by default, though - when I build with python 3.9 I get the same error as you.

parkerlreed, 2020-12-16:
Yeah Arch fully made the switch.

I'll see if I can get a 3.8 venv going and test there. Thanks.

parkerlreed, 2020-12-16:
Ok so pipenv --version 3.8 got me going

One thing that wasn't in the inkcut dependency chain is pyside2. Had to pip install that after incut in the virtual environment.

We're alive! (Dark Qt theme be damned though)

[image]

parkerlreed, 2020-12-16:
After all that my Cameo V2 isn't even working. The supported devices list shows the basic Cameo but it's not in the list in Inkcut. Only shows the Cameo 4.

raboof, 2020-12-16:
> it's not in the list

If your cutter is not in the list, there's still a good chance that it works when 'manually' configuring the right settings for it without starting from a preset (or starting from the basic Cameo preset perhaps)

parkerlreed, 2020-12-16:
That's my issue. It only has the 4 and not the basic.

[image]

I'll try manually configuring it.

parkerlreed, 2020-12-16:
Aha I missed the Cups step. Currently Im trying to find the "Generic Raw Queue Driver". Closest I have is Generic Text only

frmdstryr, 2020-12-16:
The original problem was that `enaml==0.10.2` is not compatible with `atom == 0.6.0` or python 3.8+. I'm not sure why it didn't install the most recent version of enaml (currently 0.12.0). It says 'from enamlx' but enamlx doesn't pin any versions... 

```
Requirement already satisfied: enaml in /usr/lib/python3.9/site-packages (from enamlx) (0.10.2)
```

Anyways running `python -m pip install -U enaml` should fix the original issue if anyone else sees it again.

Maybe the setup script needs to start installing different enaml versions based on the python version.

parkerlreed, 2020-12-16:
It seems it may have already been satisfied system wide and that version is out of date

```
python-enaml 0.10.2-00
```

That lagging behind plus bytecode having issues on 3.9 seem to be the main takeaways. Thanks for the help. I'll close this one out and maybe shout at bytecode.

closed 2020-12-16 by parkerlreed

parkerlreed, 2020-12-16:
FWIW enaml 0.12 also seems to fix the bytecode issue. I'll have to check with Arch and see why this is so far behind