inkcut/inkcut #304: PI driver broken?

grantemsley, 2021-03-22:
I want to implement my own driver to work with my CNC machine. The documentation says to look at the included pi driver for an example, so I tried loading that by choosing the Vinylmark R2Rv1 as my device.

As soon as I choose that, I get the error `TypeError: The 'model' member on the 'ConfigView' object must be of type 'PiConfig'. Got object of type 'DeviceConfig' instead`, and after that it's impossible to open the device settings again without going into the .config directory and deleting all the existing settings.

frmdstryr, 2021-03-27:
Yes it is fairly old. Should be removed.

That is for using an hbridge stepper driver (ST L298N) directly using gpio.  Python is too slow to get clean signals so I'd recommend using a dedicated ic.

grantemsley, 2021-03-28:
In that case is there any functional example of how to write new drivers?

frmdstryr, 2021-03-30:
If you really need a custom driver, the pi driver is the only example that I know of. 

The error sounds like a bug in the settings UI (in https://github.com/inkcut/inkcut/blob/master/inkcut/ui/settings.enaml#L105), it should probably use `Include` instead of `MappedView` and do some more generic checking.

closed 2022-08-25 by grantemsley