inkcut/inkcut #305: Sorry, an error prevented Inkcut from starting

amercer86, 2021-03-27:
Hey gang - following instructions here.

I get an error when opening InkCut. Fresh install of Rpi OS, Fresh install of Inkcut and its dependencies.

`plugin = manifest.factory()
  File "/home/pi/.local/lib/python3.7/site-packages/inkcut/preview/manifest.enaml", line 24, in plugin_factory
    from .plugin import PreviewPlugin
  File "/home/pi/.local/lib/python3.7/site-packages/inkcut/preview/plugin.py", line 12, in 
    import pyqtgraph as pg
  File "/home/pi/.local/lib/python3.7/site-packages/pyqtgraph/__init__.py", line 13, in 
    from .Qt import QtGui, mkQApp
  File "/home/pi/.local/lib/python3.7/site-packages/pyqtgraph/Qt.py", line 137, in 
    from PyQt5 import QtGui, QtCore, QtWidgets, sip, uic
ImportError: cannot import name 'sip' from 'PyQt5' (/usr/lib/python3/dist-packages/PyQt5/__init__.py) `

Thanks Friends. 

[inkcut.txt](https://github.com/inkcut/inkcut/files/6215142/inkcut.txt)
[crash.txt](https://github.com/inkcut/inkcut/files/6215143/crash.txt)

raboof, 2021-03-27:
I don't see this problem on NixOS, but various versions are newer, so that might not be saying much.

> ImportError: cannot import name 'sip' from 'PyQt5' (/usr/lib/python3/dist-packages/PyQt5/__init__.py)

This *sounds* like something to do with how your OS installed PyQt... is Rpi OS still based on debian? https://answers.ros.org/question/220572/importerror-no-module-named-sip/ mentions installing a `python-sip` package. There's also https://stackoverflow.com/questions/55087069/no-module-named-pyqt5-sip but I'm not sure that applies to your system.

frmdstryr, 2021-03-27:
https://www.riverbankcomputing.com/static/Docs/PyQt5/incompatibilities.html#pyqt-v5-11

Go to the site-packages folder and copy the sip module into the PyQt5 folder.  Or modify pyqtgraph to import sip directly.

amercer86, 2021-03-27:
@raboof Unfortunately installing sip made no difference.

@frmdstryr I don't have a sip module in my site-packages folder. Unless it goes by some other name? I download the sip package and dropped the contents into both my site-packages and the PyQt5 folder. Still no change. Tbh, not sure I'm doing this correctly though.

I've uninstalled and re-installed everything but the OS.

raboof, 2021-03-28:
> https://www.riverbankcomputing.com/static/Docs/PyQt5/incompatibilities.html#pyqt-v5-11

The error suggests pyqtgraph is already importing `sip` from `PyQt5`, so that should work with PyQt 5.11 and later, right? The output says `Qt: 5.11.3` so that suggests indeed Qt 5.11 is used - or might there be a mismatch there?

> Unfortunately installing sip made no difference

Hmm, perhaps python3-pyqt5.sip ? (https://packages.debian.org/sid/python3-pyqt5.sip)

amercer86, 2021-03-28:
Thanks @raboof  - I don't really know what to do with this information though.

This morning, I started completely over. I flashed the latest RaspiOS (full) on my B+, updated and upgraded, installed an FTP server, installed inkscape, and then lastly installed inkcut.

But I still get the exact same error - I am a little baffled that I'm the only one having this problem since its a completely vanilla installation.

frmdstryr, 2021-03-28:
Does `locate sip.cpython` show anything?

amercer86, 2021-03-28:
Sorry @frmdstryr. Is locate a search utility? I don't think its built into RpiOS.

I did do a search though and was unable to find sip.cpython

raboof, 2021-03-29:
> Thanks @raboof - I don't really know what to do with this information though.

Does `apt-get install python3-pyqt5.sip` do something? (as said, I'm not familiar with RPiOS, not sure if they're still Debian-based)

raboof, 2021-03-29:
> I am a little baffled that I'm the only one having this problem since its a completely vanilla installation.

Apparently, few people have been using inkcut on RPiOS, and those that do have either not solved this problem without (AFAICT) contacting us, or solved it without (AFAICT) contacting us.

So thanks for starting this conversation!

amercer86, 2021-03-29:
Yes, RpiOS is debian (for now). 

> Does apt-get install python3-pyqt5.sip do something?

No - package not found. 

Without knowing how all of this ties together, I'm a little suspicious of InkScape as its the only "non-standard" part of this installation - and I can see it rides on a slew of python dependencies. I'm going to start over again and omit inkscape entirely. I'll report back.

frmdstryr, 2021-03-29:
Can also try `sudo python -m pip install PyQt5`

amercer86, 2021-03-29:
> Can also try sudo python -m pip install PyQt5

No luck - but it does state that it couldn't find a version that satisfies the requirement PyQt5. Is that a hint?

[image]

frmdstryr, 2021-03-29:
What about the same command with  `python3`

amercer86, 2021-03-29:
No luck there either.

[image]

frmdstryr, 2021-03-29:
Sorry I meant like this: `sudo python3 -m pip install PyQt5`

amercer86, 2021-03-29:
@frmdstryr - Haha, yeah. I saw my mistake right after I replied. But alas - 

[image]

brentrasmussen, 2021-03-30:
I am getting this exact same error. Linux Mint 20.1.

-----
plugin = manifest.factory()
  File "/home/brentrasmussen/.local/lib/python3.8/site-packages/inkcut/preview/manifest.enaml", line 24, in plugin_factory
    from .plugin import PreviewPlugin
  File "/home/brentrasmussen/.local/lib/python3.8/site-packages/inkcut/preview/plugin.py", line 12, in 
    import pyqtgraph as pg
  File "/home/brentrasmussen/.local/lib/python3.8/site-packages/pyqtgraph/__init__.py", line 13, in 
    from .Qt import QtGui, mkQApp
  File "/home/brentrasmussen/.local/lib/python3.8/site-packages/pyqtgraph/Qt.py", line 137, in 
    from PyQt5 import QtGui, QtCore, QtWidgets, sip, uic
ImportError: cannot import name 'sip' from 'PyQt5' (/usr/lib/python3/dist-packages/PyQt5/__
-----

[inkcut.txt](https://github.com/inkcut/inkcut/files/6230592/inkcut.txt)

(my "crash.txt" file is empty.)

I already tried the suggestions in this thread, but they have not worked for me either. Thank you in advance for any additional assistance.

-Brent Rasmussen
[image]

frmdstryr, 2021-03-30:
Does `sudo pip3 install PyQt5-sip` work at all?

brentrasmussen, 2021-03-30:
Hi frmdstryr - no, sorry. It installed correctly (the "sudo pip3 install PyQt5-sip" command I mean), but I get exactly the same error as before when I try and start Inkcut.

frmdstryr, 2021-03-30:
Is  any file with `sip` somewhere in `/home/brentrasmussen/.local/lib/python3.8/site-packages` or `/usr/lib/python3/dist-packages`?

brentrasmussen, 2021-03-30:
In "/home/brentrasmussen/.local/lib/python3.8/site-packages" I have:

```
/home/brentrasmussen/.local/lib/python3.8/site-packages/twisted/protocols/__pycache__/sip.cpython-38.pyc
/home/brentrasmussen/.local/lib/python3.8/site-packages/twisted/protocols/sip.py
/home/brentrasmussen/.local/lib/python3.8/site-packages/twisted/test/__pycache__/test_sip.cpython-38.pyc
/home/brentrasmussen/.local/lib/python3.8/site-packages/twisted/test/test_sip.py
```
[image]

And in "/usr/lib/python3/dist-packages" I have:

```
/usr/lib/python3/dist-packages/sip-4.19.21.dist-info
/usr/lib/python3/dist-packages/sip.cpython-38-x86_64-linux-gnu.so
/usr/lib/python3/dist-packages/sip.pyi
/usr/lib/python3/dist-packages/__pycache__/sipconfig.cpython-38.pyc
/usr/lib/python3/dist-packages/sipconfig.py
/usr/lib/python3/dist-packages/__pycache__/sipconfig_nd8.cpython-38.pyc
/usr/lib/python3/dist-packages/sipconfig_nd8.py
```
[image]

frmdstryr, 2021-03-30:
Try copying `/usr/lib/python3/dist-packages/sip.cpython-38-x86_64-linux-gnu.so` into `/usr/lib/python3/dist-packages/PyQt5/`

brentrasmussen, 2021-03-30:
OK, did that and I have a NEW error! Must be getting somewhere. :) Here you go:

```
px, py = [v.length() if v is not None else 0 for v in self.childGroup.pixelVectors()]
  File "/home/brentrasmussen/.local/lib/python3.8/site-packages/pyqtgraph/graphicsItems/GraphicsItem.py", line 211, in pixelVectors
    dt = self.deviceTransform()
  File "/home/brentrasmussen/.local/lib/python3.8/site-packages/pyqtgraph/graphicsItems/GraphicsItem.py", line 130, in deviceTransform
    view = self.getViewWidget()
  File "/home/brentrasmussen/.local/lib/python3.8/site-packages/pyqtgraph/graphicsItems/GraphicsItem.py", line 87, in getViewWidget
    if v is not None and not isQObjectAlive(v):
  File "/home/brentrasmussen/.local/lib/python3.8/site-packages/pyqtgraph/Qt.py", line 315, in isQObjectAlive
    return not sip.isdeleted(obj)
TypeError: isdeleted() argument 1 must be sip.simplewrapper, not PlotWidget 
```
[image]

amercer86, 2021-04-27:
For giggles, I just tried on an 8gb Rpi 4. **Exact same issue.** I think its time to remove raspberry pi instructions from the install page. They don't work.

amercer86, 2021-05-04:
Attempted the steps outlined [here](https://github.com/inkcut/inkcut/issues/309#issuecomment-831266884) in: #309 

Fresh install of Rpi OS and updated. 

Build-essential was already current version, qtcreator installed without any errors - as did the rest of the inkcut installation. The same errors are still preventing Inkcut from starting. No changes.

jrperk, 2021-05-04:
Agreed.  I did a fresh install of Rpi OS and it did not work.  Went back with Ubuntu Desktop 21.04 (from raspberrypi.org) and it works with the instructions outlined in #309.  I give up.  I am going to use Ubuntu.

veego, 2021-05-05:
I solved by installing: `pip3 install pyqt5==5.14.0`

amercer86, 2021-05-05:
@veego Did you do this prior to the inkcut installation? I'm trying to do it after the inkcut install but its failing:

[image]

veego, 2021-05-05:
I did it after, but I'm on Debian10. First uninstalled `pip3 uninstall pyqt5`.

amercer86, 2021-05-05:
Ah, yes. I'm using RPiOS on a 3B+. Unfortunately still - the error persists.

amercer86, 2021-06-14:
*Crickets*

Has anybody successfully installed Inkcut on a Pi since May?

amercer86, 2021-06-23:
Closing issue w/o resolution

closed 2021-06-23 by amercer86

coogle, 2021-09-05:
I figured out how to fix this and at least get the app running again. This is on  a Raspberry Pi 3 latest Raspbian:

```
$ sudo apt-get install python3-pip python3-pyqt5 python3-setuptools libcups2-dev python3-pyqt5.qtsvg
$ sudo apt-get install libatlas-base-dev
$ pip3 install inkcut
$ pip3 install pyqtgraph==0.11.1
```

The problem is that `pyqtgraph` starting at version 0.12 deprecated support for Qt < 5.12. Inkcut, however, didn't seem to mind running on 0.11.1.