inkcut/inkcut #309: Inkcut does not start on Kubuntu 20.04 LTS

reparator, 2021-04-29:
Good Morning,
Inkcut does not start after updating to Kubuntu 20.04 LTS.
Error:
```
exec(code, module.__dict__)
  File "/home/stefan/.local/lib/python3.8/site-packages/inkcut/device/view.enaml", line 24, in 
    from inkcut.preview.plot_view import PlotView
  File "/home/stefan/.local/lib/python3.8/site-packages/inkcut/preview/plot_view.py", line 23, in 
    from pyqtgraph.widgets.PlotWidget import PlotWidget
  File "/home/stefan/.local/lib/python3.8/site-packages/pyqtgraph/__init__.py", line 13, in 
    from .Qt import QtGui, mkQApp
  File "/home/stefan/.local/lib/python3.8/site-packages/pyqtgraph/Qt.py", line 137, in 
    from PyQt5 import QtGui, QtCore, QtWidgets, sip, uic
ImportError: cannot import name 'sip' from 'PyQt5' (/usr/lib/python3/dist-packages/PyQt5/__init__.py) 
```
Looks like the same issue as in #https://github.com/inkcut/inkcut/issues/305

crash.txt is empty,
Here is inkcut.txt from the logs:
```
`2021-04-29 11:44:34,004 | INFO    | inkcut | ========================================
2021-04-29 11:44:34,004 | INFO    | inkcut | Inkcut launched
2021-04-29 11:44:34,004 | INFO    | inkcut | Version: 2.1.3
2021-04-29 11:44:34,004 | INFO    | inkcut | Python: 3.8.5 (default, Jan 27 2021, 15:41:15) 
[GCC 9.3.0]
2021-04-29 11:44:34,006 | INFO    | inkcut | System: uname_result(system='Linux', node='CMWorkstation', release='5.4.0-72-generic', version='#80-Ubuntu SMP Mon Apr 12 17:35:00 UTC 2021', machine='x86_64', processor='x86_64')
2021-04-29 11:44:34,006 | INFO    | inkcut | Executable: /usr/bin/python3
2021-04-29 11:44:34,006 | INFO    | inkcut | Args: ['/home/stefan/bin/inkcut']
2021-04-29 11:44:34,015 | INFO    | inkcut | Qt: 5.12.8 Api:
2021-04-29 11:44:34,015 | INFO    | inkcut | ========================================
2021-04-29 11:44:34,240 | DEBUG   | inkcut | Starting plugin 'inkcut.core'
2021-04-29 11:44:34,241 | DEBUG   | inkcut | Inkcut loaded.
2021-04-29 11:44:34,244 | DEBUG   | inkcut | CLI | Commands loaded
2021-04-29 11:44:34,245 | DEBUG   | inkcut | CLI | No cli command was given.
2021-04-29 11:44:34,246 | WARNING | inkcut | Setting locale: de_DE
2021-04-29 11:44:34,561 | DEBUG   | inkcut | CLI | Commands loaded
2021-04-29 11:44:34,562 | DEBUG   | inkcut | CLI | No cli command was given.
2021-04-29 11:44:34,573 | INFO    | inkcut | Loading plugin inkcut.joystick
2021-04-29 11:44:34,575 | DEBUG   | inkcut | Starting plugin 'inkcut.joystick'
2021-04-29 11:44:34,699 | INFO    | inkcut | Loading plugin inkcut.device
2021-04-29 11:44:34,752 | DEBUG   | inkcut | Starting plugin 'inkcut.device'
2021-04-29 11:44:34,759 | ERROR   | inkcut | Traceback (most recent call last):
  File "/home/stefan/.local/lib/python3.8/site-packages/inkcut/core/plugin.py", line 80, in start_default_workspace
    ui.select_workspace('inkcut.workspace')
  File "/home/stefan/.local/lib/python3.8/site-packages/enaml/workbench/ui/ui_plugin.py", line 157, in select_workspace
    new_workspace.start()
  File "/home/stefan/.local/lib/python3.8/site-packages/inkcut/ui/workspace.py", line 46, in start
    self.workbench.get_plugin('inkcut.ui')
  File "/home/stefan/.local/lib/python3.8/site-packages/enaml/workbench/workbench.py", line 151, in get_plugin
    plugin.start()
  File "/home/stefan/.local/lib/python3.8/site-packages/inkcut/ui/plugin.py", line 91, in start
    self._refresh_dock_items()
  File "/home/stefan/.local/lib/python3.8/site-packages/inkcut/ui/plugin.py", line 194, in _refresh_dock_items
    DockItem = declaration.factory()
  File "/home/stefan/.local/lib/python3.8/site-packages/inkcut/device/manifest.enaml", line 65, in device_progress_dock_item_factory
    from .view import DeviceProgressDockItem
  File "/home/stefan/.local/lib/python3.8/site-packages/enaml/core/import_hooks.py", line 159, in exec_module
    exec(code, module.__dict__)
  File "/home/stefan/.local/lib/python3.8/site-packages/inkcut/device/view.enaml", line 24, in 
    from inkcut.preview.plot_view import PlotView
  File "/home/stefan/.local/lib/python3.8/site-packages/inkcut/preview/plot_view.py", line 23, in <module>
    from pyqtgraph.widgets.PlotWidget import PlotWidget
  File "/home/stefan/.local/lib/python3.8/site-packages/pyqtgraph/__init__.py", line 13, in <module>
    from .Qt import QtGui, mkQApp
  File "/home/stefan/.local/lib/python3.8/site-packages/pyqtgraph/Qt.py", line 137, in <module>
    from PyQt5 import QtGui, QtCore, QtWidgets, sip, uic
ImportError: cannot import name 'sip' from 'PyQt5' (/usr/lib/python3/dist-packages/PyQt5/__init__.py)

2021-04-29 11:44:34,759 | CRITICAL | inkcut | 
<html>
<h3>Sorry, an error prevented Inkcut from starting</h3>
<p>
This is either a bug or a required library may be missing on your system.
As always you can request help on the forums at
<a href="https://inkcut.org">https://inkcut.org</a>.
</p>

<p>
First try to restart the application. If the error persists you can try
clearing the saved settings in <a href="/home/stefan/.config/inkcut">/home/stefan/.config/inkcut</a>
and restart the application again</a>.
</p>

<p>
If that still doesn't work please report a new issue at
<a href="https://github.com/codelv/inkcut">https://github.com/codelv/inkcut</a>
with the following error:
</p>

<pre>exec(code, module.__dict__)
  File "/home/stefan/.local/lib/python3.8/site-packages/inkcut/device/view.enaml", line 24, in 
    from inkcut.preview.plot_view import PlotView
  File "/home/stefan/.local/lib/python3.8/site-packages/inkcut/preview/plot_view.py", line 23, in <module>
    from pyqtgraph.widgets.PlotWidget import PlotWidget
  File "/home/stefan/.local/lib/python3.8/site-packages/pyqtgraph/__init__.py", line 13, in <module>
    from .Qt import QtGui, mkQApp
  File "/home/stefan/.local/lib/python3.8/site-packages/pyqtgraph/Qt.py", line 137, in <module>
    from PyQt5 import QtGui, QtCore, QtWidgets, sip, uic
ImportError: cannot import name 'sip' from 'PyQt5' (/usr/lib/python3/dist-packages/PyQt5/__init__.py)</pre>

<p> and include the logs from <a href="/home/stefan/.config/inkcut/logs">/home/stefan/.config/inkcut/logs</a>.
</p>
</html>

`
```
Any ideas to get it working?
Regards Stefan

jrperk, 2021-04-29:
I had similar issues on RPI 4 8GB with latest Ubuntu 64 bit desktop and with latest Raspberry Pi 32 bit OS.  I had this working on a headless Pi 3 B with Raspbian (created about a year ago) it still works.  Something has changed.

reparator, 2021-04-29:
> Something has changed.
Yes! But it must not be only a matter of inkcut. I assume it has something to do with pathes and python versions. The missing file (sip) is there in different places and I think inkcut does not find the right one.  But I am not python and QT expert enough to solve the issue myself.

jrperk, 2021-04-29:
I tried manually installing python 3.9 buy i still couldn't get qtconsole installed without errors.

reparator, 2021-04-29:
If I try to import in python3 console the result is:
```
~$ python3
Python 3.8.5 (default, Jan 27 2021, 15:41:15) 
[GCC 9.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from PyQt5 import sip
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: cannot import name 'sip' from 'PyQt5' (/usr/lib/python3/dist-packages/PyQt5/__init__.py)
>>> import sip
>>> 
```
So sip can be imported but not from PyQt5. Any idea?

jrperk, 2021-05-03:
I think I figured this out.  Fresh install of Ubuntu on RPI4.  
Things weren't compiling and installing correctly.  I had to add build-essential before following the installations instructions.  I also added qtcreator but I'm not sure it was really required.  

The install process that worked for me was:  

sudo apt update
sudo apt-get install build-essential
sudo apt-get install qtcreator
sudo apt-get install python3-pip python3-pyqt5 python3-setuptools libcups2-dev python3-pyqt5.qtsvg
pip3 install inkcut

I also added the .local/bin to the path so the inkcut command would work: (note use your username instead of jrp)
export PATH="/home/jrp/.local/bin:$PATH"

jrperk, 2021-05-04:
Update:  Didn't work on Raspberry Pi OS.  Didn't work on Ubuntu 20.10.  Did work on Ubuntu 21.04.  Unfortunately, 21.04 uses Wayland and that didn't work with the Raspberry Pi Element 14 7-inch touchscreen.

comteharbour, 2021-05-05:
Tried solution on Ubuntu 20.04. It doesn't work.

veego, 2021-05-05:
Same problem on Debian 10.

manu-p, 2021-05-07:
Hi, same problem on Ubuntu 20.04 and same workaround as explained by @veego. Thanks!

reparator, 2021-05-11:
Unfortunatly it does not work here:
```
2021-05-11 10:34:42,645 | INFO    | inkcut | ========================================
2021-05-11 10:34:42,645 | INFO    | inkcut | Inkcut launched
2021-05-11 10:34:42,645 | INFO    | inkcut | Version: 2.1.3
2021-05-11 10:34:42,645 | INFO    | inkcut | Python: 3.8.5 (default, Jan 27 2021, 15:41:15) 
[GCC 9.3.0]
2021-05-11 10:34:42,647 | INFO    | inkcut | System: uname_result(system='Linux', node='CMWorkstation', release='5.4.0-72-generic', version='#80-Ubuntu SMP Mon Apr 12 17:35:00 UTC 2021', machine='x86_64', processor='x86_64')
2021-05-11 10:34:42,647 | INFO    | inkcut | Executable: /usr/bin/python3
2021-05-11 10:34:42,647 | INFO    | inkcut | Args: ['./inkcut']
2021-05-11 10:34:42,663 | ERROR   | inkcut | No Qt bindings could be found
Traceback (most recent call last):
  File "/home/stefan/.local/lib/python3.8/site-packages/qtpy/__init__.py", line 204, in <module>
    from PySide import __version__ as PYSIDE_VERSION  # analysis:ignore
ModuleNotFoundError: No module named 'PySide'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/stefan/.local/lib/python3.8/site-packages/inkcut/app.py", line 78, in log_debug_info
    from qtpy import QT_VERSION
  File "/home/stefan/.local/lib/python3.8/site-packages/qtpy/__init__.py", line 210, in <module>
    raise PythonQtError('No Qt bindings could be found')
qtpy.PythonQtError: No Qt bindings could be found
2021-05-11 10:34:42,663 | INFO    | inkcut | ========================================
2021-05-11 10:34:42,665 | ERROR   | inkcut | Traceback (most recent call last):
  File "/home/stefan/.local/lib/python3.8/site-packages/qtpy/__init__.py", line 204, in <module>
    from PySide import __version__ as PYSIDE_VERSION  # analysis:ignore
ModuleNotFoundError: No module named 'PySide'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/stefan/.local/lib/python3.8/site-packages/inkcut/app.py", line 97, in main
    from inkcut.core.workbench import InkcutWorkbench
  File "/home/stefan/.local/lib/python3.8/site-packages/inkcut/core/workbench.py", line 17, in <module>
    enamlx.install()
  File "/home/stefan/.local/lib/python3.8/site-packages/enamlx/__init__.py", line 8, in install
    from enamlx.qt import qt_factories
  File "/home/stefan/.local/lib/python3.8/site-packages/enamlx/qt/qt_factories.py", line 8, in <module>
    from enaml.qt import qt_factories
  File "/home/stefan/.local/lib/python3.8/site-packages/enaml/qt/__init__.py", line 8, in <module>
    from qtpy import API as QT_API, PYQT5_API, PYSIDE2_API, QT_VERSION
  File "/home/stefan/.local/lib/python3.8/site-packages/qtpy/__init__.py", line 210, in <module>
    raise PythonQtError('No Qt bindings could be found')
qtpy.PythonQtError: No Qt bindings could be found
```
And the disadvantage of changing the pyqt5 lib to a lower version is, that some other modules will not work anymore. e.g. freecad module draft stops by not finding the right qt-version.
The error comes from inkcut and must be solved there.
Regards

clint70, 2021-06-17:
Has this been fixed... trying to install on raspbian (latest) running on RPi 4 with 2GB.  Have tried all the potential solutions here... but with no success

pi@piBCNC:~/Downloads/inkcut-2.1.3 $ inkcut
Logging to /home/pi/.config/inkcut/logs/inkcut.txt
2021-06-17 04:01:10,440 | INFO    | inkcut | ========================================
2021-06-17 04:01:10,441 | INFO    | inkcut | Inkcut launched
2021-06-17 04:01:10,441 | INFO    | inkcut | Version: 2.1.3
2021-06-17 04:01:10,441 | INFO    | inkcut | Python: 3.7.3 (default, Jan 22 2021, 20:04:44) 
[GCC 8.3.0]
2021-06-17 04:01:10,448 | INFO    | inkcut | System: uname_result(system='Linux', node='piBCNC', release='5.10.17-v7l+', version='#1421 SMP Thu May 27 14:00:13 BST 2021', machine='armv7l', processor='')
2021-06-17 04:01:10,448 | INFO    | inkcut | Executable: /usr/bin/python3
2021-06-17 04:01:10,448 | INFO    | inkcut | Args: ['/home/pi/.local/bin/inkcut']
2021-06-17 04:01:10,492 | INFO    | inkcut | Qt: 5.11.3 Api:
2021-06-17 04:01:10,493 | INFO    | inkcut | ========================================
QXcbConnection: Failed to initialize XRandr
qt5ct: using qt5ct plugin
:0: UserWarning: You do not have a working installation of the service_identity module: 'No module named 'service_identity''.  Please install it from <https://pypi.python.org/pypi/service_identity> and make sure all of its dependencies are satisfied.  Without the service_identity module, Twisted can perform only rudimentary TLS client hostname verification.  Many valid certificate/hostname mappings may be rejected.
2021-06-17 04:01:11,784 | DEBUG   | inkcut | Starting plugin 'inkcut.core'
2021-06-17 04:01:11,785 | DEBUG   | inkcut | Inkcut loaded.
2021-06-17 04:01:11,803 | DEBUG   | inkcut | CLI | Commands loaded
2021-06-17 04:01:11,806 | DEBUG   | inkcut | CLI | No cli command was given.
2021-06-17 04:01:11,808 | WARNING | inkcut | Translations not found at /home/pi/.local/lib/python3.7/site-packages/inkcut/res/translations/en_CA
qt5ct: D-Bus global menu: no
2021-06-17 04:01:12,694 | DEBUG   | inkcut | CLI | Commands loaded
2021-06-17 04:01:12,697 | DEBUG   | inkcut | CLI | No cli command was given.
2021-06-17 04:01:12,739 | INFO    | inkcut | Loading plugin inkcut.joystick
2021-06-17 04:01:12,746 | DEBUG   | inkcut | Starting plugin 'inkcut.joystick'
2021-06-17 04:01:13,787 | INFO    | inkcut | Loading plugin inkcut.monitor
2021-06-17 04:01:13,788 | DEBUG   | inkcut | Starting plugin 'inkcut.monitor'
2021-06-17 04:01:14,325 | INFO    | inkcut | Loading plugin inkcut.console
2021-06-17 04:01:15,181 | INFO    | inkcut | Loading plugin inkcut.job
2021-06-17 04:01:15,194 | DEBUG   | inkcut | Starting plugin 'inkcut.job'
2021-06-17 04:01:15,195 | INFO    | inkcut | {'type': 'create', 'object': <inkcut.job.plugin.JobPlugin object at 0xa40bdfa8>, 'name': 'job', 'value': <inkcut.job.models.Job object at 0xa403e2d8>}
2021-06-17 04:01:15,222 | ERROR   | inkcut | Traceback (most recent call last):
  File "/home/pi/.local/lib/python3.7/site-packages/inkcut/core/plugin.py", line 80, in start_default_workspace
    ui.select_workspace('inkcut.workspace')
  File "/home/pi/.local/lib/python3.7/site-packages/enaml/workbench/ui/ui_plugin.py", line 157, in select_workspace
    new_workspace.start()
  File "/home/pi/.local/lib/python3.7/site-packages/inkcut/ui/workspace.py", line 46, in start
    self.workbench.get_plugin('inkcut.ui')
  File "/home/pi/.local/lib/python3.7/site-packages/enaml/workbench/workbench.py", line 151, in get_plugin
    plugin.start()
  File "/home/pi/.local/lib/python3.7/site-packages/inkcut/ui/plugin.py", line 91, in start
    self._refresh_dock_items()
  File "/home/pi/.local/lib/python3.7/site-packages/inkcut/ui/plugin.py", line 185, in _refresh_dock_items
    plugin = workbench.get_plugin(plugin_id)
  File "/home/pi/.local/lib/python3.7/site-packages/enaml/workbench/workbench.py", line 151, in get_plugin
    plugin.start()
  File "/home/pi/.local/lib/python3.7/site-packages/inkcut/job/plugin.py", line 69, in start
    if self.job.document:
  File "/home/pi/.local/lib/python3.7/site-packages/inkcut/job/plugin.py", line 53, in _default_job
    return Job(material=self.material)
  File "/home/pi/.local/lib/python3.7/site-packages/inkcut/job/plugin.py", line 161, in _observe_material
    job = self.job
  File "/home/pi/.local/lib/python3.7/site-packages/inkcut/job/plugin.py", line 175, in _refresh_preview
    preview_plugin = self.workbench.get_plugin('inkcut.preview')
  File "/home/pi/.local/lib/python3.7/site-packages/enaml/workbench/workbench.py", line 144, in get_plugin
    plugin = manifest.factory()
  File "/home/pi/.local/lib/python3.7/site-packages/inkcut/preview/manifest.enaml", line 24, in plugin_factory
    from .plugin import PreviewPlugin
  File "/home/pi/.local/lib/python3.7/site-packages/inkcut/preview/plugin.py", line 12, in <module>
    import pyqtgraph as pg
  File "/home/pi/.local/lib/python3.7/site-packages/pyqtgraph/__init__.py", line 13, in <module>
    from .Qt import QtGui, mkQApp
  File "/home/pi/.local/lib/python3.7/site-packages/pyqtgraph/Qt.py", line 137, in <module>
    from PyQt5 import QtGui, QtCore, QtWidgets, sip, uic
ImportError: cannot import name 'sip' from 'PyQt5' (/usr/lib/python3/dist-packages/PyQt5/__init__.py)

2021-06-17 04:01:15,224 | CRITICAL | inkcut | 
<html>
[... 46 lines cut ...]
    self._refresh_dock_items()
  File "/home/pi/.local/lib/python3.7/site-packages/inkcut/ui/plugin.py", line 185, in _refresh_dock_items
[... cut ...]
  File "/home/pi/.local/lib/python3.7/site-packages/inkcut/preview/plugin.py", line 12, in <module>
    import pyqtgraph as pg
  File "/home/pi/.local/lib/python3.7/site-packages/pyqtgraph/__init__.py", line 13, in <module>
    from .Qt import QtGui, mkQApp
  File "/home/pi/.local/lib/python3.7/site-packages/pyqtgraph/Qt.py", line 137, in <module>
    from PyQt5 import QtGui, QtCore, QtWidgets, sip, uic
ImportError: cannot import name 'sip' from 'PyQt5' (/usr/lib/python3/dist-packages/PyQt5/__init__.py)
Aborted

AndreiiDeD, 2021-07-28:
I found a solution to the problem!
You need to edit the file: ~/.local/lib/python3.7/site-packages/pyqtgraph/Qt.py
line 137:
```from PyQt5 import QtGui, QtCore, QtWidgets, sip, uic```
replaced by:
```
from PyQt5 import QtGui, QtCore, QtWidgets, uic
import sip
```
and install package sip if needed
`pip install sip`

TriploidTree, 2021-08-23:
> Same problem on Debian 10.
> I solved it by installing other version of package:
> `pip3 install pyqt5==5.14.0`

Same issue here on Mint 20.2 XFCE

Using the older version of Qt works, but not sure if it'll affect anything else yet.