inkcut/inkcut #311: Only cuts in small size

abbaswasim, 2021-07-14:
When I cut a drawing with InkCut. It's only able to cut the drawing in smaller sizes, if I cut it in original size or 70% of the drawing it goes and draws one or two lines and then stops. Interestingly the "live" preview is still going and I see lots of output coming out of inkcut as well until it finishes "cleanly", things like:

```
...
2021-07-14 12:40:36,138 | DEBUG   | inkcut | -> /dev/cu.usbserial-1410 | PU13143,8639;
2021-07-14 12:40:36,140 | DEBUG   | inkcut | -> /dev/cu.usbserial-1410 | PD13143,9164;
2021-07-14 12:40:36,269 | DEBUG   | inkcut | -> /dev/cu.usbserial-1410 | PU13143,9164;
2021-07-14 12:40:36,271 | DEBUG   | inkcut | -> /dev/cu.usbserial-1410 | PD13759,8095;
2021-07-14 12:40:36,575 | DEBUG   | inkcut | -> /dev/cu.usbserial-1410 | PU13759,8095;
2021-07-14 12:40:36,577 | DEBUG   | inkcut | -> /dev/cu.usbserial-1410 | PD13759,7115;
2021-07-14 12:40:36,820 | DEBUG   | inkcut | -> /dev/cu.usbserial-1410 | PU13759,7115;
2021-07-14 12:40:36,822 | DEBUG   | inkcut | -> /dev/cu.usbserial-1410 | PD12600,5106;
2021-07-14 12:40:37,391 | DEBUG   | inkcut | -> /dev/cu.usbserial-1410 | PU0,0;
2021-07-14 12:40:37,392 | DEBUG   | inkcut | device | finish
2021-07-14 12:40:37,393 | DEBUG   | inkcut | device | disconnect
2021-07-14 12:40:37,393 | DEBUG   | inkcut | -- /dev/cu.usbserial-1410 | closed by request
2021-07-14 12:40:37,397 | DEBUG   | inkcut | -- /dev/cu.usbserial-1410 | dropped: [Failure instance: Traceback (failure with no frames): <class 'twisted.internet.error.ConnectionDone'>: Connection was closed cleanly.
]
```

I am using InkCut with the generic driver. Here is a picture of it easily manage to draw the smaller drawing but not the larger one.

[image]

Here is the drawing I am talking about in zip and a picture of it.
[Fez_Door_TEST_50X50-01.svg.zip](https://github.com/inkcut/inkcut/files/6815751/Fez_Door_TEST_50X50-01.svg.zip)

[image]

How can I debug this? what might be causing this issue?

I am on macOS  Big Sur 11.4.

claespol, 2021-07-19:
Look at Issue #306 for eventual solution to your problem.
Rgds

abbaswasim, 2021-07-19:
Thanks, that explains it.

abbaswasim, 2021-07-22:
> Look at Issue #306 for eventual solution to your problem.
> Rgds

I am not sure if this solution has solved my problem completely. At the moment it's not that the plotting stops halfway through but it jumps from part of the drawing to another part and never comes back to finish off. It's like something get overridden half way through the drawing process. My input file size is way more than the plotter internal buffer which seems to be 16kb. I see a few chunks of 16kb data copied correctly but after a few such cycles things go south.

Any ideas what might be causing this?

claespol, 2021-08-11:
5x 3x c4    =                .      r.  g       b v y f  ,   b.   z   d
s,

abbaswasim, 2021-08-29:
So I didn’t solve my inkcut issues and I can’t directly help but what I can recommend is to try this cable that I bought which has definitely made a big difference. 

Search 
 ```
UGREEN USB Serial Cable USB to RS232 DB9 9 pin Converter Cable for Connecting Cisco Routers and Switches,Celestron Telescope Nexstar hand Controller, Extron Crestron Products with PL2303 Chipset (2m)
```
On Amazon. This comes with a driver that you would need to install. I haven’t tried this with Inkcut but this has helped me in a different way. 

The issue that I noticed with my setup was that the USB-A cable I was using wasn’t communicating any of the hardware flow control signals. I verified this with another serial coms applications. When I would send commands to the serial port it won’t signal any ready etc coms  back. When I changed to the actual serial cable with a driver I could then see the hardware flow control signals in the application. If at this point I try my setup with InKcut it might work as well but when I was trying different options and got frustrated I ended up writing my own plotter software. So I never went back to inkcut. 

Everything works fine for me now. Unfortunately I can’t be sure if it’s because of the new cable and driver or my bespoke plotter software.

My software is very specific to my plotter, an unbranded Chinese clone of something, so I can’t share or release it yet either. It also only supports DMPL.

abbaswasim, 2026-02-24:
Nope. As I said in the earlier comment. I wrote my own plotter SW never went back to InkCut.