inkcut/inkcut #315: Problem with Inkcut

arratos, 2021-08-13:
i can not install Inkcut 
Can anyone help me?
plugin = manifest.factory()
  File "/home/arratos/.local/lib/python3.8/site-packages/inkcut/preview/manifest.enaml", line 24, in plugin_factory
    from .plugin import PreviewPlugin
  File "/home/arratos/.local/lib/python3.8/site-packages/inkcut/preview/plugin.py", line 12, in 
    import pyqtgraph as pg
  File "/home/arratos/.local/lib/python3.8/site-packages/pyqtgraph/__init__.py", line 13, in 
    from .Qt import QtCore, QtGui, mkQApp
  File "/home/arratos/.local/lib/python3.8/site-packages/pyqtgraph/Qt/__init__.py", line 155, in 
    from PyQt5 import sip, uic
ImportError: cannot import name 'sip' from 'PyQt5' (/usr/lib/python3/dist-packages/PyQt5/__init__.py)   

2021-08-13 14:05:54,380 | INFO    | inkcut | Inkcut launched
2021-08-13 14:05:54,380 | INFO    | inkcut | Version: 2.1.3
2021-08-13 14:05:54,380 | INFO    | inkcut | Python: 3.8.10 (default, Jun  2 2021, 10:49:15) 
[GCC 9.4.0]
2021-08-13 14:05:54,383 | INFO    | inkcut | System: uname_result(system='Linux', node='arratos', release='5.8.0-63-generic', version='#71~20.04.1-Ubuntu SMP Thu Jul 15 17:46:08 UTC 2021', machine='x86_64', processor='x86_64')
2021-08-13 14:05:54,383 | INFO    | inkcut | Executable: /usr/bin/python3
2021-08-13 14:05:54,383 | INFO    | inkcut | Args: ['/home/arratos/.local/bin/inkcut']
2021-08-13 14:05:54,396 | INFO    | inkcut | Qt: 5.12.8 Api:
2021-08-13 14:05:54,396 | INFO    | inkcut | ========================================
2021-08-13 14:05:54,789 | DEBUG   | inkcut | Starting plugin 'inkcut.core'
2021-08-13 14:05:54,789 | DEBUG   | inkcut | Inkcut loaded.
2021-08-13 14:05:54,797 | DEBUG   | inkcut | CLI | Commands loaded
2021-08-13 14:05:54,798 | DEBUG   | inkcut | CLI | No cli command was given.
2021-08-13 14:05:54,809 | WARNING | inkcut | Translations not found at /home/arratos/.local/lib/python3.8/site-packages/inkcut/res/translations/el_GR
2021-08-13 14:05:55,053 | DEBUG   | inkcut | CLI | Commands loaded
2021-08-13 14:05:55,054 | DEBUG   | inkcut | CLI | No cli command was given.
2021-08-13 14:05:55,061 | INFO    | inkcut | Loading plugin inkcut.console
2021-08-13 14:05:55,378 | INFO    | inkcut | Loading plugin inkcut.device
2021-08-13 14:05:55,439 | DEBUG   | inkcut | Starting plugin 'inkcut.device'
2021-08-13 14:05:55,463 | ERROR   | inkcut | Traceback (most recent call last):
  File "/home/arratos/.local/lib/python3.8/site-packages/inkcut/core/plugin.py", line 80, in start_default_workspace
    ui.select_workspace('inkcut.workspace')
  File "/home/arratos/.local/lib/python3.8/site-packages/enaml/workbench/ui/ui_plugin.py", line 157, in select_workspace
    new_workspace.start()
  File "/home/arratos/.local/lib/python3.8/site-packages/inkcut/ui/workspace.py", line 46, in start
    self.workbench.get_plugin('inkcut.ui')
  File "/home/arratos/.local/lib/python3.8/site-packages/enaml/workbench/workbench.py", line 151, in get_plugin
    plugin.start()
  File "/home/arratos/.local/lib/python3.8/site-packages/inkcut/ui/plugin.py", line 91, in start
    self._refresh_dock_items()
  File "/home/arratos/.local/lib/python3.8/site-packages/inkcut/ui/plugin.py", line 194, in _refresh_dock_items
    DockItem = declaration.factory()
  File "/home/arratos/.local/lib/python3.8/site-packages/inkcut/device/manifest.enaml", line 65, in device_progress_dock_item_factory
    from .view import DeviceProgressDockItem
  File "/home/arratos/.local/lib/python3.8/site-packages/enaml/core/import_hooks.py", line 159, in exec_module
    exec(code, module.__dict__)
  File "/home/arratos/.local/lib/python3.8/site-packages/inkcut/device/view.enaml", line 24, in 
    from inkcut.preview.plot_view import PlotView
  File "/home/arratos/.local/lib/python3.8/site-packages/inkcut/preview/plot_view.py", line 23, in <module>
    from pyqtgraph.widgets.PlotWidget import PlotWidget
  File "/home/arratos/.local/lib/python3.8/site-packages/pyqtgraph/__init__.py", line 13, in <module>
    from .Qt import QtCore, QtGui, mkQApp
  File "/home/arratos/.local/lib/python3.8/site-packages/pyqtgraph/Qt/__init__.py", line 155, in <module>
    from PyQt5 import sip, uic
ImportError: cannot import name 'sip' from 'PyQt5' (/usr/lib/python3/dist-packages/PyQt5/__init__.py)

2021-08-13 14:05:55,463 | CRITICAL | inkcut | 
<html>
<h3>Sorry, an error prevented Inkcut from starting</h3>
<p>
This is either a bug or a required library may be missing on your system.
As always you can request help on the forums at
<a href="https://inkcut.org">https://inkcut.org</a>.
</p>

<p>
First try to restart the application. If the error persists you can try
[... 83 lines cut ...]
First try to restart the application. If the error persists you can try
clearing the saved settings in <a href="/home/arratos/.config/inkcut">/home/arratos/.config/inkcut</a>
and restart the application again</a>.
</p>

<p>
If that still doesn't work please report a new issue at
<a href="https://github.com/codelv/inkcut">https://github.com/codelv/inkcut</a>
with the following error:
</p>

<pre>plugin = manifest.factory()
  File "/home/arratos/.local/lib/python3.8/site-packages/inkcut/preview/manifest.enaml", line 24, in plugin_factory
    from .plugin import PreviewPlugin
  File "/home/arratos/.local/lib/python3.8/site-packages/inkcut/preview/plugin.py", line 12, in <module>
    import pyqtgraph as pg
  File "/home/arratos/.local/lib/python3.8/site-packages/pyqtgraph/__init__.py", line 13, in <module>
    from .Qt import QtCore, QtGui, mkQApp
  File "/home/arratos/.local/lib/python3.8/site-packages/pyqtgraph/Qt/__init__.py", line 155, in <module>
    from PyQt5 import sip, uic
ImportError: cannot import name 'sip' from 'PyQt5' (/usr/lib/python3/dist-packages/PyQt5/__init__.py)</pre>

<p> and include the logs from <a href="/home/arratos/.config/inkcut/logs">/home/arratos/.config/inkcut/logs</a>.
</p>
</html>

z1969, 2021-08-28:
Same Problem with RPi4 and RaspiOS

arratos, 2021-10-20:
> @arratos <https://github.com/arratos>
>
> python3 -m pip install --upgrade PyQt5-sip

Thanks very much