inkcut/inkcut #320: Inkut not working on Fedora 35

Smurv, 2021-11-16:
Hi,
on Fedora 35 there is an error if I want to start Inkcut. Any idea how make it work?

**Error message:**
> Traceback (most recent call last):
  File "/home/smurv/.local/lib/python3.10/site-packages/enaml/qt/q_deferred_caller.py", line 42, in customEvent
    event.callback(*event.args, **event.kwargs)
  File "/home/smurv/.local/lib/python3.10/site-packages/inkcut/core/plugin.py", line 80, in start_default_workspace
    ui.select_workspace('inkcut.workspace')
  File "/home/smurv/.local/lib/python3.10/site-packages/enaml/workbench/ui/ui_plugin.py", line 157, in select_workspace
    new_workspace.start()
  File "/home/smurv/.local/lib/python3.10/site-packages/inkcut/ui/workspace.py", line 46, in start
    self.workbench.get_plugin('inkcut.ui')
  File "/home/smurv/.local/lib/python3.10/site-packages/enaml/workbench/workbench.py", line 151, in get_plugin
    plugin.start()
  File "/home/smurv/.local/lib/python3.10/site-packages/inkcut/ui/plugin.py", line 91, in start
    self._refresh_dock_items()
  File "/home/smurv/.local/lib/python3.10/site-packages/inkcut/ui/plugin.py", line 194, in _refresh_dock_items
    DockItem = declaration.factory()
  File "/home/smurv/.local/lib/python3.10/site-packages/inkcut/preview/manifest.enaml", line 30, in preview_factory
    from .view import PreviewDockItem
ImportError: cannot import name 'PreviewDockItem' from 'inkcut.preview.view' (/home/smurv/.local/lib/python3.10/site-packages/inkcut/preview/view.enaml)

Log file attached.
[inkcut.txt](https://github.com/inkcut/inkcut/files/7544860/inkcut.txt)

Manu3l0us, 2021-11-16:
I see the same issue on my side with FC35. Could it be a python 3.10 compatibility issue with inkcut or enaml?

frmdstryr, 2021-11-16:
> Could it be a python 3.10 compatibility issue with inkcut or enaml?

Yes, the current release of enaml doesn't yet work with 3.10, try moving to 3.9 or upgrade to enaml to 0.14.rc1 (from github).

Smurv, 2021-11-16:
> Yes, the current release of enaml doesn't yet work with 3.10, try moving to 3.9 or upgrade to enaml to 0.14.rc1 (from github).

Thank for your reply. I tried to install github version but pip give me some errors. I'm not expert in python. I will try virtual enviroment later.

vmario89, 2021-12-25:
having similar issue

```
2021-12-25 20:45:11,794 | INFO    | inkcut | ========================================
2021-12-25 20:45:11,794 | INFO    | inkcut | Inkcut launched
2021-12-25 20:45:11,794 | INFO    | inkcut | Version: 2.1.3
2021-12-25 20:45:11,794 | INFO    | inkcut | Python: 3.10.0 (default, Oct  4 2021, 00:00:00) [GCC 11.2.1 20210728 (Red Hat 11.2.1-1)]
2021-12-25 20:45:11,794 | INFO    | inkcut | System: uname_result(system='Linux', node='rock', release='5.15.8-200.fc35.x86_64', version='#1 SMP Tue Dec 14 14:26:01 UTC 2021', machine='x86_64')
2021-12-25 20:45:11,794 | INFO    | inkcut | Executable: /usr/bin/python3
2021-12-25 20:45:11,794 | INFO    | inkcut | Args: ['/usr/local/bin/inkcut']
2021-12-25 20:45:11,814 | INFO    | inkcut | Qt: 5.15.2 Api:
2021-12-25 20:45:11,814 | INFO    | inkcut | ========================================
2021-12-25 20:45:12,172 | DEBUG   | inkcut | Starting plugin 'inkcut.core'
2021-12-25 20:45:12,172 | DEBUG   | inkcut | Inkcut loaded.
2021-12-25 20:45:12,177 | DEBUG   | inkcut | CLI | Commands loaded
2021-12-25 20:45:12,179 | DEBUG   | inkcut | CLI | No cli command was given.
2021-12-25 20:45:12,180 | WARNING | inkcut | Setting locale: de_DE
2021-12-25 20:45:12,460 | DEBUG   | inkcut | CLI | Commands loaded
2021-12-25 20:45:12,461 | DEBUG   | inkcut | CLI | No cli command was given.
2021-12-25 20:45:12,470 | INFO    | inkcut | Loading plugin inkcut.device
2021-12-25 20:45:12,533 | DEBUG   | inkcut | Starting plugin 'inkcut.device'
2021-12-25 20:45:12,733 | INFO    | inkcut | Loading plugin inkcut.job
2021-12-25 20:45:12,736 | DEBUG   | inkcut | Starting plugin 'inkcut.job'
2021-12-25 20:45:12,736 | INFO    | inkcut | {'type': 'create', 'object': <inkcut.job.plugin.JobPlugin object at 0x7f14c0752c00>, 'name': 'job', 'value': <inkcut.job.models.Job object at 0x7f14c065d980>}
2021-12-25 20:45:12,737 | DEBUG   | inkcut | Starting plugin 'inkcut.preview'
2021-12-25 20:45:12,737 | INFO    | inkcut | Saving state due to change: {'type': 'update', 'object': <inkcut.device.plugin.DevicePlugin object at 0x7f14ec50afc0>, 'name': 'devices', 'oldvalue': [], 'value': [<inkcut.device.plugin.Device object at 0x7f14cee7b880>]}
2021-12-25 20:45:12,737 | INFO    | inkcut | {}
2021-12-25 20:45:12,738 | INFO    | inkcut | Saving state due to change: {'type': 'request'}
2021-12-25 20:45:12,741 | INFO    | inkcut | device | 
2021-12-25 20:45:12,743 | INFO    | inkcut | {}
2021-12-25 20:45:12,744 | INFO    | inkcut | Saving state due to change: {'type': 'request'}
2021-12-25 20:45:12,746 | INFO    | inkcut | Saving state due to change: {'type': 'request'}
2021-12-25 20:45:12,747 | INFO    | inkcut | {'type': 'create', 'object': <inkcut.job.plugin.JobPlugin object at 0x7f14c0752c00>, 'name': 'job', 'value': <inkcut.job.models.Job object at 0x7f14e5b9ea40>}
2021-12-25 20:45:12,747 | INFO    | inkcut | Saving state due to change: {'type': 'request'}
2021-12-25 20:45:12,900 | INFO    | inkcut | Loading plugin inkcut.job
2021-12-25 20:45:12,900 | INFO    | inkcut | Loading plugin inkcut.job
2021-12-25 20:45:12,900 | INFO    | inkcut | Loading plugin inkcut.job
2021-12-25 20:45:12,901 | INFO    | inkcut | Loading plugin inkcut.job
2021-12-25 20:45:12,901 | INFO    | inkcut | Loading plugin inkcut.joystick
2021-12-25 20:45:12,901 | DEBUG   | inkcut | Starting plugin 'inkcut.joystick'
2021-12-25 20:45:12,938 | INFO    | inkcut | Loading plugin inkcut.preview
2021-12-25 20:45:12,971 | ERROR   | inkcut | Traceback (most recent call last):
  File "/home/himbeere/.local/lib/python3.10/site-packages/inkcut/core/plugin.py", line 80, in start_default_workspace
    ui.select_workspace('inkcut.workspace')
  File "/home/himbeere/.local/lib/python3.10/site-packages/enaml/workbench/ui/ui_plugin.py", line 157, in select_workspace
    new_workspace.start()
  File "/home/himbeere/.local/lib/python3.10/site-packages/inkcut/ui/workspace.py", line 46, in start
    self.workbench.get_plugin('inkcut.ui')
  File "/home/himbeere/.local/lib/python3.10/site-packages/enaml/workbench/workbench.py", line 151, in get_plugin
    plugin.start()
  File "/home/himbeere/.local/lib/python3.10/site-packages/inkcut/ui/plugin.py", line 91, in start
    self._refresh_dock_items()
  File "/home/himbeere/.local/lib/python3.10/site-packages/inkcut/ui/plugin.py", line 194, in _refresh_dock_items
    DockItem = declaration.factory()
  File "/home/himbeere/.local/lib/python3.10/site-packages/inkcut/preview/manifest.enaml", line 30, in preview_factory
    from .view import PreviewDockItem
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "/home/himbeere/.local/lib/python3.10/site-packages/enaml/core/import_hooks.py", line 140, in exec_module
    code, _ = self.get_code()
  File "/home/himbeere/.local/lib/python3.10/site-packages/enaml/core/import_hooks.py", line 401, in get_code
    return self.compile_code()
  File "/home/himbeere/.local/lib/python3.10/site-packages/enaml/core/import_hooks.py", line 366, in compile_code
    code = EnamlCompiler.compile(ast, file_info.src_path)
  File "/home/himbeere/.local/lib/python3.10/site-packages/enaml/core/enaml_compiler.py", line 178, in compile
    return compiler.visit(node)
  File "/home/himbeere/.local/lib/python3.10/site-packages/enaml/core/enaml_ast.py", line 359, in visit
[... cut ...]
  File "/home/himbeere/.local/lib/python3.10/site-packages/bytecode/cfg.py", line 391, in from_bytecode
    if not block[-1].is_final():
  File "/home/himbeere/.local/lib/python3.10/site-packages/bytecode/cfg.py", line 44, in __getitem__
    value = super().__getitem__(index)
IndexError: list index out of range</pre>

<p> and include the logs from <a href="/home/himbeere/.config/inkcut/logs">/home/himbeere/.config/inkcut/logs</a>.
</p>
</html>

```

btw i needed to install inkcut with sudo. otherwise the inkcut command does not appear on path :-O any better idea?

cdedwards, 2022-01-26:
after installing 3.9 python and installing pip3, pyqt5, I was able to get rid of the error and inkcut started as it should. My other issue was a viewport problem. I designed my layout in  lightburn (cause it was handy) off of the cutting area and then exported it. I had a viewport which had a negative number in it. This caused inkcut (and me) to refuse to load the image svg. When I hand edited the file and replaced the negative with a 0, the image loaded and was even the proper size.

frmdstryr, 2022-02-09:
This is now fixed. Anyone on python 3.10 can use `pip install -U enaml` to upgrade to `0.14.1`

closed 2022-02-09 by frmdstryr