inkcut/inkcut #323: Inkcut on windows reporting `ImportError: cannot import name 'PYSIDE_API' from 'qtpy'` just after installation

mofosyne, 2022-01-07:
Log showed this error line `ImportError: cannot import name 'PYSIDE_API' from 'qtpy' (C:\Users\Brian\AppData\Roaming\Python\Python39\site-packages\qtpy\__init__.py)`

as shown below from `C:\Users\Brian\.config\inkcut\logs`

```
2022-01-07 15:25:17,583 | INFO    | inkcut | ========================================
2022-01-07 15:25:17,585 | INFO    | inkcut | Inkcut launched
2022-01-07 15:25:17,585 | INFO    | inkcut | Version: 2.1.3
2022-01-07 15:25:17,586 | INFO    | inkcut | Python: 3.9.1 (tags/v3.9.1:1e5d33e, Dec  7 2020, 17:08:21) [MSC v.1927 64 bit (AMD64)]
2022-01-07 15:25:17,587 | INFO    | inkcut | System: uname_result(system='Windows', node='DESKTOP-G93D7TB', release='10', version='10.0.19041', machine='AMD64')
2022-01-07 15:25:17,587 | INFO    | inkcut | Executable: c:\program files\python39\python.exe
2022-01-07 15:25:17,588 | INFO    | inkcut | Args: ['C:\\Users\\Brian\\AppData\\Roaming\\Python\\Python39\\Scripts\\inkcut']
2022-01-07 15:25:17,611 | INFO    | inkcut | Qt: 5.15.2 Api:
2022-01-07 15:25:17,613 | INFO    | inkcut | ========================================
2022-01-07 15:25:18,223 | DEBUG   | inkcut | Starting plugin 'inkcut.core'
2022-01-07 15:25:18,223 | DEBUG   | inkcut | Inkcut loaded.
2022-01-07 15:25:18,238 | DEBUG   | inkcut | CLI | Commands loaded
2022-01-07 15:25:18,239 | DEBUG   | inkcut | CLI | No cli command was given.
2022-01-07 15:25:18,241 | WARNING | inkcut | Translations not found at C:\Users\Brian\AppData\Roaming\Python\Python39\site-packages\inkcut\res\translations\en_AU
2022-01-07 15:25:18,977 | DEBUG   | inkcut | CLI | Commands loaded
2022-01-07 15:25:18,980 | DEBUG   | inkcut | CLI | No cli command was given.
2022-01-07 15:25:18,992 | INFO    | inkcut | Loading plugin inkcut.console
2022-01-07 15:25:19,540 | INFO    | inkcut | Loading plugin inkcut.joystick
2022-01-07 15:25:19,544 | DEBUG   | inkcut | Starting plugin 'inkcut.joystick'
2022-01-07 15:25:19,599 | INFO    | inkcut | Loading plugin inkcut.device
2022-01-07 15:25:19,730 | DEBUG   | inkcut | Starting plugin 'inkcut.device'
2022-01-07 15:25:20,049 | INFO    | inkcut | Loading plugin inkcut.job
2022-01-07 15:25:20,067 | DEBUG   | inkcut | Starting plugin 'inkcut.job'
2022-01-07 15:25:20,068 | INFO    | inkcut | {'type': 'create', 'object': <inkcut.job.plugin.JobPlugin object at 0x0000022DEDF860C0>, 'name': 'job', 'value': <inkcut.job.models.Job object at 0x0000022DEE40BD40>}
2022-01-07 15:25:20,069 | DEBUG   | inkcut | Starting plugin 'inkcut.preview'
2022-01-07 15:25:20,070 | INFO    | inkcut | Saving state due to change: {'type': 'update', 'object': <inkcut.device.plugin.DevicePlugin object at 0x0000022DD7A838C0>, 'name': 'devices', 'oldvalue': [], 'value': [<inkcut.device.plugin.Device object at 0x0000022DEE3FF440>]}
2022-01-07 15:25:20,070 | INFO    | inkcut | {}
2022-01-07 15:25:20,071 | INFO    | inkcut | Saving state due to change: {'type': 'request'}
2022-01-07 15:25:20,131 | INFO    | inkcut | device | 
2022-01-07 15:25:20,139 | INFO    | inkcut | {}
2022-01-07 15:25:20,140 | INFO    | inkcut | Saving state due to change: {'type': 'request'}
2022-01-07 15:25:20,143 | INFO    | inkcut | Saving state due to change: {'type': 'request'}
2022-01-07 15:25:20,146 | INFO    | inkcut | {'type': 'create', 'object': <inkcut.job.plugin.JobPlugin object at 0x0000022DEDF860C0>, 'name': 'job', 'value': <inkcut.job.models.Job object at 0x0000022DD7B81F00>}
2022-01-07 15:25:20,146 | INFO    | inkcut | Saving state due to change: {'type': 'request'}
2022-01-07 15:25:20,268 | INFO    | inkcut | Loading plugin inkcut.job
2022-01-07 15:25:20,269 | INFO    | inkcut | Loading plugin inkcut.job
2022-01-07 15:25:20,270 | INFO    | inkcut | Loading plugin inkcut.job
2022-01-07 15:25:20,270 | INFO    | inkcut | Loading plugin inkcut.job
2022-01-07 15:25:20,271 | INFO    | inkcut | Loading plugin inkcut.preview
2022-01-07 15:25:20,284 | INFO    | inkcut | Loading plugin inkcut.monitor
2022-01-07 15:25:20,287 | DEBUG   | inkcut | Starting plugin 'inkcut.monitor'
2022-01-07 15:25:20,312 | DEBUG   | inkcut | Updating dock items: [<inkcut.console.view.ConsoleDockItem object at 0x0000022DD68CF440>, <inkcut.joystick.view.JoystickDockItem object at 0x0000022DD7A83AC0>, <inkcut.device.view.DeviceProgressDockItem object at 0x0000022DD7B73F80>, <inkcut.job.view.GraphicDockItem object at 0x0000022DD7B81E40>, <inkcut.job.view.MaterialDockItem object at 0x0000022DEE487C40>, <inkcut.job.view.WeedlinesDockItem object at 0x0000022DEE4A7D40>, <inkcut.job.view.LayersDockItem object at 0x0000022DEE4EF600>, <inkcut.job.view.JobHistoryDockItem object at 0x0000022DEE48E080>, <inkcut.preview.view.PreviewDockItem object at 0x0000022DEE4DA740>, <inkcut.monitor.view.MonitorDockItem object at 0x0000022DEE4FCD00>]
2022-01-07 15:25:20,380 | DEBUG   | inkcut | Updating settings pages: [<inkcut.ui.extensions.SettingsPage object at 0x0000022DD68BFAC0>, <inkcut.ui.extensions.SettingsPage object at 0x0000022DD2D48580>, <inkcut.ui.extensions.SettingsPage object at 0x0000022DD68BFF80>]
2022-01-07 15:25:20,381 | DEBUG   | inkcut | Starting plugin 'inkcut.ui'
2022-01-07 15:25:20,434 | ERROR   | inkcut | Traceback (most recent call last):
  File "C:\Users\Brian\AppData\Roaming\Python\Python39\site-packages\inkcut\core\plugin.py", line 80, in start_default_workspace
    ui.select_workspace('inkcut.workspace')
  File "C:\Users\Brian\AppData\Roaming\Python\Python39\site-packages\enaml\workbench\ui\ui_plugin.py", line 158, in select_workspace
    self._model.workspace = new_workspace
  File "C:\Users\Brian\AppData\Roaming\Python\Python39\site-packages\enaml\core\standard_tracer.py", line 55, in __call__
    engine.update(owner, self.name)
  File "C:\Users\Brian\AppData\Roaming\Python\Python39\site-packages\enaml\core\expression_engine.py", line 241, in update
    setattr(owner, name, pair.reader(owner, name))
  File "C:\Users\Brian\AppData\Roaming\Python\Python39\site-packages\enaml\core\include.py", line 85, in _observe_objects
    self.parent.insert_children(self, newvalue)
  File "C:\Users\Brian\AppData\Roaming\Python\Python39\site-packages\enaml\core\object.py", line 228, in insert_children
    child_added(child)
  File "C:\Users\Brian\AppData\Roaming\Python\Python39\site-packages\enaml\widgets\toolkit_object.py", line 182, in child_added
    super(ToolkitObject, self).child_added(child)
  File "C:\Users\Brian\AppData\Roaming\Python\Python39\site-packages\enaml\styling.py", line 343, in child_added
    super(Stylable, self).child_added(child)
  File "C:\Users\Brian\AppData\Roaming\Python\Python39\site-packages\enaml\core\declarative.py", line 140, in child_added
    child.initialize()
  File "C:\Users\Brian\AppData\Roaming\Python\Python39\site-packages\enaml\widgets\toolkit_object.py", line 160, in initialize
    super(ToolkitObject, self).initialize()
[... 59 lines cut ...]
    from enamlx.qt.qt_abstract_item_view import (
  File "C:\Users\Brian\AppData\Roaming\Python\Python39\site-packages\enamlx\qt\qt_abstract_item_view.py", line 14, in <module>
    from qtpy import QT_API, PYSIDE_API, PYQT4_API
ImportError: cannot import name 'PYSIDE_API' from 'qtpy' (C:\Users\Brian\AppData\Roaming\Python\Python39\site-packages\qtpy\__init__.py)</pre>

<p> and include the logs from <a href="C:\Users\Brian/.config/inkcut\logs">C:\Users\Brian/.config/inkcut\logs</a>.
</p>
</html>
```

I was following this install instruction from [here](https://www.codelv.com/projects/inkcut/download/)

```
Next install the dependencies

pip3 install pyqt5

Then install inkcut

pip3 install inkcut

Start it by running

inkcut
```

mofosyne, 2022-01-07:
Found to be duplicate of https://github.com/inkcut/inkcut/issues/321

frmdstryr, 2022-01-07:
This was fixed in enamlx

closed 2022-01-07 by frmdstryr