inkcut/inkcut #325: inkcut does not start

tom0099, 2022-01-22:
Hello,
I use Manjaro Linux. Inkcut does not start.
Do you have any ideas?
Regards Tom...

return bc_code.to_code()
  File "/home/thorn/.local/lib/python3.10/site-packages/bytecode/bytecode.py", line 204, in to_code
    stacksize = self.compute_stacksize(check_pre_and_post=check_pre_and_post)
  File "/home/thorn/.local/lib/python3.10/site-packages/bytecode/bytecode.py", line 195, in compute_stacksize
    cfg = _bytecode.ControlFlowGraph.from_bytecode(self)
  File "/home/thorn/.local/lib/python3.10/site-packages/bytecode/cfg.py", line 391, in from_bytecode
    if not block[-1].is_final():
  File "/home/thorn/.local/lib/python3.10/site-packages/bytecode/cfg.py", line 44, in __getitem__
    value = super().__getitem__(index)
IndexError: list index out of range

[inkcut.txt](https://github.com/inkcut/inkcut/files/7918916/inkcut.txt)

cdedwards, 2022-01-25:
I'm getting this same error on Archlinux.

frmdstryr, 2022-01-25:
Try using python 3.9. There have been other [issues reported](https://github.com/inkcut/inkcut/issues/320) with python 3.10

closed 2022-01-25 by frmdstryr