inkcut/inkcut #341: Opening job from stdin only works half the time

raboof, 2022-07-29:
Since 748ab4157f87afec37dadd715094e87d02c9c739, passing in a document via stdin (like `inkcut open - < foo.svg`) only works every other attempt.

This notably impacts opening documents via the Inkscape extension

frmdstryr, 2022-09-05:
Sounds like the something with saved settings is preventing it from re-opening.

I'm curious if you've tried using copy paste? Just copy in inkscape and paste in inkcut (or right click and open from clipboard).

swfboschma, 2025-03-16:
I installed Inkcut 2.15 today. I noticed that on Windows the Plugin will only work every other attempt.

When I edited the .config/inkcut/inkcut.job.json file, by removing the '-' for document, it would start

As a solution to get it started every time, I added the next code in class Job(Model):

    def __getstate__(self):
        """ Exclude any members from the state where document does point to stdin - """
        state = super(Job, self).__getstate__()
        if state["document"] == "-": # Stdin, would crash the Plugin every second time
            state["document"] = ''
        return state

frmdstryr, 2025-04-12:
Thanks for the fix @swfboschma

closed 2025-04-12 by frmdstryr