inkcut/inkcut #349: OSX 12.6 (monterey) installed but won't run

pxdiaz, 2022-09-28:
Inkcut launches but comes back with an error. Inkscape 1.2, Cameo 4 Pro cutter

What do I dooooooo?

```
philipdiaz@Philips-MBP / % inkcut
Logging to /Users/philipdiaz/.config/inkcut/logs/inkcut.txt
2022-09-27 18:38:51,270 | INFO    | inkcut | ========================================
2022-09-27 18:38:51,270 | INFO    | inkcut | Inkcut launched
2022-09-27 18:38:51,270 | INFO    | inkcut | Version: 2.1.5
2022-09-27 18:38:51,271 | INFO    | inkcut | Python: 3.9.6 (default, Aug  5 2022, 15:21:02) 
[Clang 14.0.0 (clang-1400.0.29.102)]
2022-09-27 18:38:51,271 | INFO    | inkcut | System: uname_result(system='Darwin', node='Philips-MBP.attlocal.net', release='21.6.0', version='Darwin Kernel Version 21.6.0: Mon Aug 22 20:17:10 PDT 2022; root:xnu-8020.140.49~2/RELEASE_X86_64', machine='x86_64')
2022-09-27 18:38:51,271 | INFO    | inkcut | Executable: /Library/Developer/CommandLineTools/usr/bin/python3
2022-09-27 18:38:51,271 | INFO    | inkcut | Args: ['/Users/philipdiaz/Library/Python/3.9/bin/inkcut']
2022-09-27 18:38:51,280 | INFO    | inkcut | Qt: None Api:
2022-09-27 18:38:51,280 | INFO    | inkcut | ========================================
2022-09-27 18:38:51,295 | ERROR   | inkcut | Traceback (most recent call last):
  File "/Users/philipdiaz/Library/Python/3.9/lib/python/site-packages/inkcut/app.py", line 97, in main
    from inkcut.core.workbench import InkcutWorkbench
  File "/Users/philipdiaz/Library/Python/3.9/lib/python/site-packages/inkcut/core/workbench.py", line 22, in <module>
    from enaml.qt import QT_API, QtCore, QtWidgets, QtGui
  File "/Users/philipdiaz/Library/Python/3.9/lib/python/site-packages/enaml/qt/QtCore.py", line 12, in <module>
    __version_info__ = tuple(map(int, QT_VERSION.split('.')))
AttributeError: 'NoneType' object has no attribute 'split'

Traceback (most recent call last):
  File "/Users/philipdiaz/Library/Python/3.9/bin/inkcut", line 8, in <module>
    sys.exit(main())
  File "/Users/philipdiaz/Library/Python/3.9/lib/python/site-packages/inkcut/app.py", line 97, in main
    from inkcut.core.workbench import InkcutWorkbench
  File "/Users/philipdiaz/Library/Python/3.9/lib/python/site-packages/inkcut/core/workbench.py", line 22, in <module>
    from enaml.qt import QT_API, QtCore, QtWidgets, QtGui
  File "/Users/philipdiaz/Library/Python/3.9/lib/python/site-packages/enaml/qt/QtCore.py", line 12, in <module>
    __version_info__ = tuple(map(int, QT_VERSION.split('.')))
AttributeError:` 'NoneType' object has no attribute 'split'

```

andy83-projects, 2023-04-17:
i have Monterey with similar but different issue

Traceback (most recent call last):
  File "~/Library/Application Support/org.inkscape.Inkscape/config/inkscape/extensions/inkcut_open.py", line 82, in <module>
    InkscapeInkcutPlugin().run()
  File "/Applications/Inkscape.app/Contents/Resources/share/inkscape/extensions/inkex/base.py", line 231, in run
    self.save_raw(self.effect())
  File "~/Library/Application Support/org.inkscape.Inkscape/config/inkscape/extensions/inkcut_open.py", line 67, in effect
    p = subprocess.Popen(cmd,
  File "/Applications/Inkscape.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/subprocess.py", line 969, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/Applications/Inkscape.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/subprocess.py", line 1845, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: 'inkcut'

karliss, 2023-04-17:
@andy83-projects Can you create a separate issue? It has nothing in common with this issue and discussing it here will just make extra noise for pxdiaz.