inkcut/inkcut #354: On inkcut command : cannot import name 'PYQT6_API' from 'qtpy'

To2NY74, 2022-10-16:
Hello,
I have a problem running inkcut on a freshly installed PI3B+ using the latest Raspberry pi OS (Raspbian GNU/Linux 11 (bullseye)):

The installation seemed ok but when i try to run inkcut via the line command I have this error:
```
Logging to /home/pi/.config/inkcut/logs/inkcut.txt
2022-10-16 17:24:19,586 | INFO    | inkcut | ========================================
2022-10-16 17:24:19,587 | INFO    | inkcut | Inkcut launched
2022-10-16 17:24:19,590 | INFO    | inkcut | Version: 2.1.5
2022-10-16 17:24:19,591 | INFO    | inkcut | Python: 3.9.2 (default, Mar 12 2021, 04:06:34) 
[GCC 10.2.1 20210110]
2022-10-16 17:24:19,592 | INFO    | inkcut | System: uname_result(system='Linux', node='raspberrypi', release='5.15.61-v7+', version='#1579 SMP Fri Aug 26 11:10:59 BST 2022', machine='armv7l')
2022-10-16 17:24:19,593 | INFO    | inkcut | Executable: /usr/bin/python3
2022-10-16 17:24:19,593 | INFO    | inkcut | Args: ['/home/pi/.local/lib/python3.9/site-packages/inkcut/__main__.py']
2022-10-16 17:24:19,680 | INFO    | inkcut | Qt: 5.15.2 Api:
2022-10-16 17:24:19,681 | INFO    | inkcut | ========================================
2022-10-16 17:24:19,694 | ERROR   | inkcut | Traceback (most recent call last):
  File "/home/pi/.local/lib/python3.9/site-packages/inkcut/app.py", line 97, in main
    from inkcut.core.workbench import InkcutWorkbench
  File "/home/pi/.local/lib/python3.9/site-packages/inkcut/core/workbench.py", line 17, in <module>
    enamlx.install()
  File "/home/pi/.local/lib/python3.9/site-packages/enamlx/__init__.py", line 6, in install
    from enamlx.qt import qt_factories  # noqa: F401
  File "/home/pi/.local/lib/python3.9/site-packages/enamlx/qt/qt_factories.py", line 8, in <module>
    from enaml.qt import qt_factories
  File "/home/pi/.local/lib/python3.9/site-packages/enaml/qt/__init__.py", line 8, in <module>
    from qtpy import (
ImportError: cannot import name 'PYQT6_API' from 'qtpy' (/usr/lib/python3/dist-packages/qtpy/__init__.py)

Traceback (most recent call last):
  File "/usr/lib/python3.9/runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.9/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/home/pi/.local/lib/python3.9/site-packages/inkcut/__main__.py", line 16, in <module>
    main()
  File "/home/pi/.local/lib/python3.9/site-packages/inkcut/app.py", line 97, in main
    from inkcut.core.workbench import InkcutWorkbench
  File "/home/pi/.local/lib/python3.9/site-packages/inkcut/core/workbench.py", line 17, in <module>
    enamlx.install()
  File "/home/pi/.local/lib/python3.9/site-packages/enamlx/__init__.py", line 6, in install
    from enamlx.qt import qt_factories  # noqa: F401
  File "/home/pi/.local/lib/python3.9/site-packages/enamlx/qt/qt_factories.py", line 8, in <module>
    from enaml.qt import qt_factories
  File "/home/pi/.local/lib/python3.9/site-packages/enaml/qt/__init__.py", line 8, in <module>
    from qtpy import (
ImportError: cannot import name 'PYQT6_API' from 'qtpy' (/usr/lib/python3/dist-packages/qtpy/__init__.py)
```

Here is my pip freeze :

```
pip freeze
appdirs==1.4.4
arandr==0.1.10
astroid==2.5.1
asttokens==2.0.4
atom==0.8.1
attrs==22.1.0
Automat==20.2.0
automationhat==0.2.0
backcall==0.2.0
beautifulsoup4==4.9.3
blinker==1.4
blinkt==0.1.2
buttonshim==0.0.2
bytecode==0.13.0
Cap1xxx==0.1.3
certifi==2020.6.20
chardet==4.0.0
click==7.1.2
colorama==0.4.4
colorzero==1.1
constantly==15.1.0
cryptography==3.3.2
cupshelpers==1.0
cycler==0.10.0
dbus-python==1.2.16
decorator==4.4.2
distro==1.5.0
docutils==0.16
drumhat==0.1.0
enaml==0.15.2
enamlx==0.6.2
envirophat==1.0.0
ExplorerHAT==0.4.2
Flask==1.1.2
fourletterphat==0.1.0
gpiozero==1.6.2
guizero==1.1.1
html5lib==1.1
hyperlink==21.0.0
idna==2.10
incremental==22.10.0
inkcut==2.1.5
ipykernel==5.4.3
ipython==7.20.0
ipython-genutils==0.2.0
isort==5.6.4
itsdangerous==1.1.0
jedi==0.18.0
Jinja2==2.11.3
jsonpickle==2.2.0
jupyter-client==6.1.11
jupyter-core==4.7.1
kiwisolver==1.3.1
lazy-object-proxy==0.0.0
logilab-common==1.8.1
lxml==4.6.3
MarkupSafe==1.1.1
matplotlib==3.3.4
mccabe==0.6.1
microdotphat==0.2.1
mock==4.0.3
mote==0.0.4
motephat==0.0.3
mypy==0.812
mypy-extensions==0.4.3
nudatus==0.0.5
numpy==1.23.4
oauthlib==3.1.0
packaging==21.3
pantilthat==0.0.7
parso==0.8.1
pbr==5.5.0
pexpect==4.8.0
pgzero==1.2
phatbeat==0.1.1
pianohat==0.1.0
picamera==1.13
picamera2==0.3.3
pickleshare==0.7.5
pidng==4.0.9
piexif==1.1.3
piglow==1.2.5
pigpio==1.78
Pillow==8.1.2
ply==3.11
prompt-toolkit==3.0.14
psutil==5.8.0
pycairo==1.16.2
pycodestyle==2.6.0
pycups==2.0.1
pyflakes==2.2.0
pygame==1.9.6
Pygments==2.7.1
PyGObject==3.38.0
pyinotify==0.9.6
PyJWT==1.7.1
pylint==2.7.2
PyOpenGL==3.1.5
pyOpenSSL==20.0.1
pyparsing==2.4.7
PyQt5==5.15.2
PyQt5-sip==12.11.0
pyqtgraph==0.13.1
pyserial==3.5
pysmbc==1.0.23
python-apt==2.2.1
python-dateutil==2.8.1
python-prctl==1.7
pyzmq==20.0.0
QScintilla==2.11.6
qt-reactor==0.6
qtconsole==5.0.2
QtPy==1.9.0
rainbowhat==0.1.0
reportlab==3.5.59
requests==2.25.1
requests-oauthlib==1.0.0
responses==0.12.1
roman==2.0.0
RPi.GPIO==0.7.0
RTIMULib==7.2.1
scour==0.38.2
scrollphat==0.0.7
scrollphathd==1.2.1
semver==2.10.2
Send2Trash==1.6.0b1
sense-emu==1.2
sense-hat==2.4.0
simplejpeg==1.6.4
simplejson==3.17.2
sip==6.7.2
six==1.16.0
skywriter==0.0.7
sn3218==1.2.7
soupsieve==2.2.1
spidev==3.5
ssh-import-id==5.10
thonny==3.3.14
toml==0.10.1
tornado==6.1
touchphat==0.0.1
traitlets==5.0.5
Twisted==22.8.0
twython==3.8.2
typed-ast==1.4.2
typing-extensions==3.7.4.3
uflash==1.2.4
unicornhathd==0.0.4
urllib3==1.26.5
v4l2-python3==0.3.1
wcwidth==0.1.9
webencodings==0.5.1
Werkzeug==1.0.1
wrapt==1.12.1
zope.interface==5.5.0
```

Anyone can help ? Thank you for reading this :)

karliss, 2022-10-16:
Looks like that version of enaml is incompatible with that version of qtpy. You either need a newer qtpy or older enaml.

To2NY74, 2022-10-16:
Hello thanks for your quick answer !

I did upgrade successfully qtpy to 2.2.1 using `pip3 install --upgrade qtpy`

Then I tried inkcut again and the error message changed ! Looks like you were into something :)

I did a few things following the error messages then, something like that : 
`pip install service-identity`
`pip install -U pyopenssl`
`sudo apt-get install libatlas-base-dev`
`pip3 uninstall numpy`
`sudo apt install python3-numpy`

then it seems to work, I`ll try it on my machine some other day. Note that some of those commands lines may have not done anything, I just installed whatever was missing following the error messages.

Thanks again you led me to the right way !

closed 2022-10-16 by To2NY74