inkcut/inkcut #358: Inkcut not starting on Ubuntu

itwhiz33, 2022-10-29:
I've got rid of a lot of error by trying to workout what dependency is missing, but I'm still getting 
Sorry, an error prevented Inkcut from starting. Any ideas would be a great help.

This is either a bug or a required library may be missing on your system. As always you can request help on the forums at https://inkcut.org/. 
First try to restart the application. If the error persists you can try clearing the saved settings in [/home/itwhiz/.config/inkcut](https://github.com/home/itwhiz/.config/inkcut) and restart the application again. 
If that still doesn't work please report a new issue at https://github.com/codelv/inkcut with the following error: 
cls = resolver.resolve(name)
  File "/home/itwhiz/.local/lib/python3.8/site-packages/enaml/application.py", line 147, in resolve
    return factory()
  File "/home/itwhiz/.local/lib/python3.8/site-packages/enamlx/qt/qt_factories.py", line 132, in table_view_factory
    from .qt_table_view import QtTableView
  File "/home/itwhiz/.local/lib/python3.8/site-packages/enamlx/qt/qt_table_view.py", line 13, in 
    from enamlx.qt.qt_abstract_item_view import (
  File "/home/itwhiz/.local/lib/python3.8/site-packages/enamlx/qt/qt_abstract_item_view.py", line 14, in 
    from qtpy import QT_API, PYSIDE_API, PYQT4_API
ImportError: cannot import name 'PYSIDE_API' from 'qtpy' (/home/itwhiz/.local/lib/python3.8/site-packages/qtpy/__init__.py) 
and include the logs from [/home/itwhiz/.config/inkcut/logs](https://github.com/home/itwhiz/.config/inkcut/logs).

karliss, 2022-10-29:
I would guess that the version of enamlx and qtpy are incompatible and you probably need either newer enamlx or older qtpy.

itwhiz33, 2022-10-29:
Tried different combinations and still nothing works.
Log file
[inkcut (copy).txt](https://github.com/inkcut/inkcut/files/9894687/inkcut.copy.txt)

Also having issues on windows 10. 

May try loading onto a cleanly installed Ubuntu box tomorrow.

karliss, 2022-10-30:
> Tried different combinations and still nothing works.

What versions did you actually try? And how did you install them, which ones came from pip and which ones from ubuntu repositories?  Telling that you "different versions" means nothing to me.

itwhiz33, 2022-11-02:
I got it working from my old laptop running Ubuntu 18.04 and installing inkcut using

apt-get install python3-pip python3-pyqt5 python3-setuptools libcups2-dev python3-pyqt5.qtsvg
pip3 install inkcut

Found main PC also has update issues because of an invalid Public KEY

Need to work out now

1) Switching the origin - my cutter (LiYu SC631) has it's home point at the right hand side of the machine, inkcut has it on the left.

2) Why Inkscape 1.2.1 will not open files or save from the application. The only way I've found to open so far is by right clicking on the file and using 'Open With ....', but still can't save.
 
Thanks for all you help.

karliss, 2022-11-02:
@itwhiz33 Anything left to do in this issue, are you satisfied with it running on your old laptop or can this issue be closed, ? Unless you are willing to spend your time and help us with further investigating of what exactly happened there is not much that can be done about your initial problem.

As for rest of your issues
1, 3) You should be able to adjust under Device/Output tab. There is "Mirror X" and also output scaling options. Note that there might be few issue with the numbers and corner displayed in Preview area but it should be possible to configure it so that physical dimensions and direction matches the dimensions in Inkscape.

2) If I understood you correctly this sounds more like Inkscape issue not Inkcut. Did you install Inkscape using Flatpack or Snap package? It might have the sandboxing configured badly so that Inkscape can't freely access your filesystem only permitting to touch  files that you opened in certain way.