inkcut/inkcut #365: InkCut startup Attribute Error

DJLevel3, 2022-12-10:
On inkcut startup, I get an attribute error popup saying to report this error if it persists after deleting settings and restarting. Lo and behold it did persist, so I'm posting this error.

The message in question:

```
File "/home/dj_level_3/.local/lib/python3.10/site-packages/jupyter_client/provisioning/factory.py", line 19, in <module>
    from .provisioner_base import KernelProvisionerBase
  File "/home/dj_level_3/.local/lib/python3.10/site-packages/jupyter_client/provisioning/provisioner_base.py", line 24, in <module>
    class KernelProvisionerBase(ABC, LoggingConfigurable, metaclass=KernelProvisionerMeta):
  File "/home/dj_level_3/.local/lib/python3.10/site-packages/traitlets/traitlets.py", line 972, in __init__
    cls.setup_class(classdict)
  File "/home/dj_level_3/.local/lib/python3.10/site-packages/traitlets/traitlets.py", line 1009, in setup_class
    value = getattr(cls, name)
AttributeError: __provides__
```

Logs:
[crash.txt](https://github.com/inkcut/inkcut/files/10200432/crash.txt)
[inkcut.txt](https://github.com/inkcut/inkcut/files/10200433/inkcut.txt)

I was able to progress a little further in the application and set up my machine, but I can't find the Graphtec CE6000-60 that is on the list of supported machines.

pitatus, 2022-12-12:
@DJLevel3 I'm getting the same error, what did you do to make progress past it?

DJLevel3, 2022-12-12:
@pitatus I decided to use the free edition of Graphtec's official software since I couldn't get it to work. If I can find a way I'll switch back but for now I'm putting up with it

jhogendorn, 2022-12-12:
I've tried it with python 3.10, 3.9, 3.8. all the same error. `pip install --upgrade traitlets` doesnt help, nor trying to update jupyter_client or any of the the other intevening libs.

karliss, 2022-12-12:
Can anyone of you with this problem send:
* exact python version you used
* operating system
* List of all installed package versions obtained using  `pip list` (or `pip3 list` on certain linux distros)

pitatus, 2022-12-12:
<pre>$ python -V
Python 3.10.6</pre>

<pre>$ cat /etc/os-release 
PRETTY_NAME="Ubuntu 22.04.1 LTS"</pre>

<pre>
$ pip list
Package            Version
------------------ -------
asttokens          2.2.1
atom               0.8.2
attrs              22.1.0
Automat            22.10.0
backcall           0.2.0
bytecode           0.14.0
comm               0.1.2
constantly         15.1.0
debugpy            1.6.4
decorator          5.1.1
enaml              0.15.2
enamlx             0.6.2
entrypoints        0.4
executing          1.2.0
hyperlink          21.0.0
idna               3.4
incremental        22.10.0
inkcut             2.1.5
ipykernel          6.19.2
ipython            8.7.0
ipython-genutils   0.2.0
jedi               0.18.2
jsonpickle         3.0.0
jupyter_client     7.4.8
jupyter_core       5.1.0
kiwisolver         1.4.4
lxml               4.9.1
matplotlib-inline  0.1.6
nest-asyncio       1.5.6
numpy              1.23.5
packaging          22.0
parso              0.8.3
pexpect            4.8.0
pickleshare        0.7.5
pip                22.0.2
platformdirs       2.6.0
ply                3.11
prompt-toolkit     3.0.36
psutil             5.9.4
ptyprocess         0.7.0
pure-eval          0.2.2
pycups             2.0.1
Pygments           2.13.0
PyQt5              5.15.7
PyQt5-Qt5          5.15.2
PyQt5-sip          12.11.0
pyqtgraph          0.13.1
pyserial           3.5
PySide6            6.4.1
PySide6-Addons     6.4.1
PySide6-Essentials 6.4.1
python-dateutil    2.8.2
pyzmq              24.0.1
qt-reactor         0.6
qtconsole          5.4.0
QtPy               2.3.0
setuptools         59.6.0
shiboken6          6.4.1
six                1.16.0
stack-data         0.6.2
tornado            6.2
traitlets          5.7.0
Twisted            22.10.0
typing_extensions  4.4.0
wcwidth            0.2.5
zope.interface     5.5.2
</pre>

karliss, 2022-12-14:
Thanks @pitatus seems like I was able to find the cause by comparing list of packages you provided against what I got in a working venv. You have traitlets package 5.7.0 I had 5.7.1. When I downgraded my traitlets package from 5.7.1 to 5.7.0 I started, after upgrading it back to 5.7.1 the problem disappeared.

The changelong for 5.7.1 https://github.com/ipython/traitlets/releases/tag/v5.7.1 which was released 2 days ago contains a bugfix for what seems like probably the same error related to jupyter_client/provisioning/provisioner_base.py and  AttributeError: __provides__  .

karliss, 2022-12-14:
I guess trying to upgrade traitlets didn't help jhogendorn  because he was unlucky enough to do it few hours before bugfix got fully released . Doing it now should install the version with bugfix.

pitatus, 2022-12-16:
@karliss  That did it for me, thanks for looking into it!