inkcut/inkcut #366: Inkcut doesn't start on Fedora 37. Enaml errors, AST node line range is not valid

marvokurven, 2022-12-19:
Running Fedora 37 with the latest updates installed 2 days prior to this writing, and all dependencies should be installed also. I do however suspect some version mismatch in some of the dependencies, because every time I try to start it, it shows a lot of errors related to enaml. 
Usually, I'm able to solve these kind of problems myself, but since pyhthon probably has the most cryptic error messages of any software, and I don't know python myself, then I'm kinda lost.

Tried reinstalling inkcut, reinstalling the dependencies, and also installed more python/pyqt related packages, like pyqt5, pyqt6, pysvg. Reinstalled inkcut everytime, but it still stops at the same error:
**ValueError: AST node line range (8, 1) is not valid**
And a lot of enaml errors in the log. And due to the cryptic nature of python errors, I'm suspecting that "something" is wrong with enaml, but it doesn't say what, at least not in an understandable way for laymans who doesn't know python.

This happens whether I run inkcut from the extensions menu in inkscape, or directly from the terminal. If I run it from inkscape *with no paths selected* then it doesn't produce any errors at all, but simply halts with a message saying no paths are selected, -as is expected of course.

[Log here](https://pastebin.com/AK02bd20)

karliss, 2022-12-19:
Can you add output of `pip list` so that we know what exact versions of python packages you got installed? That helps a lot with debugging issues related to version mismatch or buggy versions of dependencies.

marvokurven, 2022-12-19:
`Package             Version
------------------- -----------
aeidon              1.11
aiodns              3.0.0
aiohttp             3.8.3
aiosignal           1.3.1
appdirs             1.4.4
argcomplete         2.0.0
asttokens           2.2.1
async-timeout       4.0.2
atom                0.8.2
attrs               22.1.0
aubio               0.4.9
Automat             22.10.0
Babel               2.10.3
backcall            0.2.0
Beaker              1.10.0
beautifulsoup4      4.11.0
binwalk             2.3.2
black               22.8.0
blivet              3.5.0
blivet-gui          2.4.0
bottle              0.12.21
Brotli              1.0.9
bytecode            0.14.0
cached-property     1.5.2
certifi             2021.10.8
cffi                1.15.1
chardet             5.1.0
charset-normalizer  2.1.0
Cheetah3            3.2.6.post1
cheroot             8.6.0
click               8.1.3
click-plugins       1.1.1
comm                0.1.2
constantly          15.1.0
contourpy           1.0.6
cryptography        37.0.2
cssselect           1.1.0
cupshelpers         1.0
cycler              0.11.0
dasbus              1.6
dbus-python         1.3.2
debugpy             1.6.4
decorator           5.1.1
defusedxml          0.7.1
distro              1.7.0
dnspython           2.2.1
enaml               0.15.2
enamlx              0.6.2
entrypoints         0.4
evdev               1.6.0
executing           1.2.0
file-magic          0.4.0
fonttools           4.37.2
frozenlist          1.3.3
fs                  2.4.11
future              0.18.2
gattlib             0.20210616
Glances             3.3.0.4
gpg                 1.17.0
gssapi              1.7.3
humanize            3.13.1
hyperlink           21.0.0
icoextract          0.1.4
idna                3.4
incremental         22.10.0
iniparse            0.5
initial-setup       0.3.95
inkcut              2.1.5
ipykernel           6.19.2
ipython             8.7.0
ipython-genutils    0.2.0
jaraco.functools    3.5.1
jedi                0.18.2
jsonpickle          3.0.1
jupyter_client      7.4.8
jupyter_core        5.1.0
kiwisolver          1.4.4
koji                1.31.0
langtable           0.0.61
libcomps            0.1.18
libvirt-python      8.6.0
lutris              0.5.11
lxml                4.9.2
Mako                1.1.4
Markdown            3.4.1
MarkupSafe          2.1.1
matplotlib          3.6.2
matplotlib-inline   0.1.6
meson               0.63.3
more-itertools      8.12.0
multidict           6.0.2
munkres             1.1.2
mutagen             1.45.1
mypy-extensions     0.4.3
natsort             7.1.1
nest-asyncio        1.5.6
nftables            0.1
ntplib              0.3.3
numpy               1.24.0
olefile             0.46
ordered-set         4.1.0
packaging           22.0
parso               0.8.3
Paste               3.5.0
pathspec            0.10.1
patool              1.12
pefile              2022.5.30
pexpect             4.8.0
pickleshare         0.7.5
pid                 2.2.3
Pillow              9.2.0
pip                 22.2.2
platformdirs        2.6.0
ply                 3.11
productmd           1.33
progressbar2        3.53.2
prompt-toolkit      3.0.36
psutil              5.9.4
ptyprocess          0.7.0
pure-eval           0.2.2
pwquality           1.4.4
pyasn1              0.4.8
pyasn1-modules      0.2.8
PyBluez             0.23
pycairo             1.21.0
pycares             4.2.2
pycollada           0.7.2
pycountry           22.3.5
pycparser           2.20
pycrypto            2.6.1
pycryptodomex       3.15.0
pycups              2.0.1
pycurl              7.45.1
pyenchant           3.2.2
pygit2              1.7.1
Pygments            2.13.0
PyGObject           3.42.2
pyinotify           0.9.6
pykickstart         3.41
PyOpenGL            3.1.5
PyOpenGL-accelerate 3.1.5
pyOpenSSL           21.0.0
pyparsing           3.0.9
pyparted            3.12.0
PyQt5               5.15.6
PyQt5-sip           12.11.0
PyQt6               6.4.0
PyQt6-sip           13.4.0
pyqtgraph           0.13.1
pyserial            3.5
PySide2             5.15.7
PySocks             1.7.1
python-augeas       1.1.0
python-dateutil     2.8.2
python-gettext      4.0
python-Levenshtein  0.12.0
python-manatools    0.0.4
python-meh          0.50
python-utils        3.1.0
pytz                2022.6
pyudev              0.23.2
pyxdg               0.27
PyYAML              6.0
pyzmq               24.0.1
qt-reactor          0.6
qtconsole           5.4.0
QtPy                2.3.0
requests            2.28.1
requests-file       1.5.1
requests-ftp        0.3.1
requests-gssapi     1.2.3
rpm                 4.18.0
rpmautospec         0.3.0
ruamel.yaml         0.17.21
ruamel.yaml.clib    0.2.6
SciPy               1.8.1
scour               0.38.1
selinux             3.4
sepolicy            3.4
service-identity    21.1.0
setools             4.4.0
setproctitle        1.2.3
setuptools          65.6.3
shiboken2           5.15.7
shiboken2-generator 5.15.7
simplejson          3.17.6
simpleline          1.9.0
six                 1.16.0
slip                0.6.4
slip.dbus           0.6.4
sos                 4.4
soupsieve           2.3.2.post1
SSSDConfig          2.8.1
stack-data          0.6.2
svg.path            4.1
svgwrite            1.4.3
systemd-python      235
Tempita             0.5.2
thrift              0.14.0
toml                0.10.2
tomli               2.0.1
tornado             6.2
traitlets           5.7.1
translate-toolkit   3.6.2
trustme             0.9.0
Twisted             22.10.0
typing_extensions   4.4.0
urllib3             1.26.12
vobject             0.9.6.1
wcwidth             0.2.5
websockets          10.3
wxPython            4.2.0
yarl                1.7.2
yt-dlp              2022.11.11
zope.event          4.5.0
zope.interface      5.5.2`

karliss, 2022-12-19:
This is just a guess, but enaml might be having some issues with python 3.11 .  Will try to install python 3.11 for testing this theory.

marvokurven, 2022-12-19:
Also, I don't know if it's relevant, but I'm running KDE Plasma 5.26.4.1 and Wayland 1.21 -Intel graphics with default driver, no Nvidia hardware in my computer.

karliss, 2022-12-19:
@marvokurven This https://github.com/nucleic/enaml/pull/514 more or less confirms that enaml (important library used by inkcut) doesn't support python3.11 yet. As a temporary workaround you can use something like pyenv to setup additional python versions without messing up your system python and everything which depends on it.