inkcut/inkcut #370: Start cutting position looks wrong

cfnz, 2023-01-31:
Hi, as seen from the screen shot, the "live" view indicates that the top left start of the capital T is starting in the wrong place.

I would have thought that the line across the top of the capital T would be horizontal.

This is using the Inkcut Generic Driver, saving to file setting.

Version is 2.1.5, Python: 3.10.6, Ubuntu 22.04

Is this expected behaviour?

[image]

karliss, 2023-01-31:
The problem is that at the start inkcut has no idea about the initial orientation of blade  which result in weird cut like that.  I am a bit annoyed by that as well. If it went horizontally it would be slightly less bed, but the problem of blade being oriented in random direction would remain resulting in the ends of cut not connecting properly (depending on how unlucky you are). There is room for slightly smarter approach which will result in better result even if initial blade orientation is unknown.

As workaround I usually add small alignment cut. And since I am too lazy to remember from which corner it will start I usually add 4 of them, one in each corner. Something like that:

[image]

cfnz, 2023-01-31:
OK, that makes sense. Thanks.

A nice (but maybe not high priority) feature would be to allow addition of an alignment cut from within Inkcut, maybe initially just by specifying position and size in a dialog. Should I add a separate Feature Request or is there just too much other stuff to do when there is a workaround from the designing tool (as you explained)?

karliss, 2023-02-01:
Please just reopen this issue. While I explained to you why it happens nothing really got fixed yet and the problem is still there with multiple potential changes that would make it less bad.