inkcut/inkcut #378: i am working on support for silver bullet, black cat, knk max, knk zing cutters

geekinesis, 2024-07-11:
I am working on support for several cutters which i have reverse engineered the proprietry hpgl code for...
Mostly its a case of different commands for pen up and pen down ( silver bullet is just U instead of PU etc)
And also some specifc commands for the laser etc. 

So i am wondering if there is any way of me adding these specific commands into the existing inkcut as it stands as a user...
Alternatively I am able to code 'plugins' to support variations for different devices, although i dont think inkcut has a plugin option to add new devices permanently with specific device commands?...

Alternatively i am currently working on my own open source cutting software in iosx and linux based on swift and flutter... and will support various devices.

geekinesis, 2024-07-11:
the force and speed should be left disabled
then in the device/job commands/before
write the following

ECN U PA;
US165;
!FS200.0;
VS165.0;

set protocol to DMPL and option to 3

that sets the speed force and up speed using the silver bullet protocols and initialises the cutter. (the first line isnt strictly necessary but i dont know what it means so put it in there for safety as thats what I see gets sent using a serial sniffer)

geekinesis, 2024-07-11:
I will upload settings for the knk max and knk zing shortly.

frmdstryr, 2024-07-19:
Feel free to add new entries here https://github.com/inkcut/inkcut/blob/master/inkcut/device/drivers/manifest.enaml