inkcut/inkcut #379: contour cutting

geekinesis, 2024-07-11:
is there a plan for contour cutting? If so is there something already in progress?
If not I would like to contribute to implementing contour cut.

frmdstryr, 2024-07-19:
There is an empty pluging I started a long time ago but never implemented https://github.com/inkcut/inkcut/tree/master/inkcut/device/contour

geekinesis, 2024-07-19:
> There is an empty plugin I started a long time ago but never implemented https://github.com/inkcut/inkcut/tree/master/inkcut/device/contour

Thank youI have already written the code to do this so just need to add an interface in line with what you have already designed. I originally wrote  the code for my own flutter app but have translated it to python as I also use ink cut for cutting. At the moment I am using the pygame library to allow the user to jog the cutting head with keyboard arrow keys. I will try and find a different way of doing that. Otherwise I’ll connect to the jog buttons you already designed. Sent from my iPhone

geekinesis, 2024-07-26:
just working on a contour cutting interface, this 
[image]
is what i have so far - it would be a seperate window but i need to make it look like your main window style... 
it works with my cutter at the moment but some cutters have different protocols/methods for contour registration so have to work on that....