inkcut/inkcut #387: InkCut not starting on Ubuntu 22.04 with pipx - No Qt bindings could be found

vladaman, 2024-11-04:
Steps:

```
sudo apt-get install python-pyqt5 python-pyqt5.qtsvg python-pyqt5.qtwebkit
pipx install inkcut
inkcut
```

error:
```
Traceback (most recent call last):
  File "/home/vlad/.local/bin/inkcut", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/home/vlad/.local/share/pipx/venvs/inkcut/lib/python3.12/site-packages/inkcut/app.py", line 97, in main
    from inkcut.core.workbench import InkcutWorkbench
  File "/home/vlad/.local/share/pipx/venvs/inkcut/lib/python3.12/site-packages/inkcut/core/workbench.py", line 17, in <module>
    enamlx.install()
  File "/home/vlad/.local/share/pipx/venvs/inkcut/lib/python3.12/site-packages/enamlx/__init__.py", line 6, in install
    from enamlx.qt import qt_factories  # noqa: F401
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/vlad/.local/share/pipx/venvs/inkcut/lib/python3.12/site-packages/enamlx/qt/qt_factories.py", line 8, in <module>
    from enaml.qt import qt_factories
  File "/home/vlad/.local/share/pipx/venvs/inkcut/lib/python3.12/site-packages/enaml/qt/__init__.py", line 8, in <module>
    from qtpy import (
  File "/home/vlad/.local/share/pipx/venvs/inkcut/lib/python3.12/site-packages/qtpy/__init__.py", line 287, in <module>
    raise QtBindingsNotFoundError from None
qtpy.QtBindingsNotFoundError: No Qt bindings could be found

```

sftwrjnke, 2024-11-24:
i had the same problem. you need to get PyQt5 into the pipx virtual environment inkcut folder which is like a container that contains all dependencies and apparently isn't looking elsewhere for missing dependencies (has a requirements.text document that it references and pulls them in; which you don't see (it is in the source code folder). option 1 'sudo apt-get install' puts PyQt5 in your /usr/lib/python3/dist-packages folder and option 2 has it's own venvs location; but 'pipx install inkcut' installs to it's own venvs folder. but didn't seem to pull in PyQt5. 

so here's what i did and inkcut opens and seems to be working but i my plotter is in storage so i haven't gotten to test it yet: 
1)  add pipx to your PATH in terminal with 'pipx ensurepath' minus the quotes if you haven't already
2) copy PyQt5 folder from  /usr/lib/python3/dist-packages, go to /home/yourusernamehere/.local/share/pipx/venvs/inkcut/lib/ open whatever python folder is there (mine's python3.12) open 'site-packges' folder and paste PyQt5
3) open terminal run 'inkcut' (minus the quotes) but it'll close if you close the terminal

to run inkcut without having to keep the terminal open go to /home/yourusernamehere/.local/share/pipx/venvs/inkcut/bin and double click inkcut file. 

make a desktop or menu launcher if so inclined.

sftwrjnke, 2024-11-24:
> Instead of messing with manually copying PyQt files you should be able to
> add them to pipx inkcut environment using `pipx inject. See
> https://github.com/pypa/pipx?tab=readme-ov-file#inject-a-package for more
> details

thanks. i will check that out. i've been using ubuntu mate desktop for a
few years and got started compiling from source just for the experience and
a bunch of programs i tried that on were fairly old synth packages so i
started trying to fix errors often finding answers, though like correct, a
little too vague for a beginner to figure out. not knowing any programming
i'd tinker around finding and updating obsolete syntax, substituting new
versions of libraries or following lucky guesses; sometimes getting them to
work, other times feeling defeated realizing what dependency hell it would
was trying to replace/update every outdated package to get something to
work. i can just imagine how time consuming that must have been back when
you had to hunt down each one manually.

cyril279, 2025-08-25:
The`--system-site-packages` option is another way to solve this issue.  
I just learned about this today, and have verified that it works.  
Previously I was symbolically linking the system site-packages to the .local/../venvs.  
```sh
pipx install git+https://github.com/codelv/inkcut.git --system-site-packages
```

from `pipx install --help`:
```sh
--system-site-packages
                        Give the virtual environment access to the
                        system site-packages dir.
```