inkcut/inkcut #397: linux mint

leresteux, 2025-04-14:
hi 
inkcut crashs on my computer 
---
If that still doesn't work please report a new issue at
<a href="https://github.com/codelv/inkcut">https://github.com/codelv/inkcut</a>
with the following error:
</p>

<pre>new_workspace.start()
  File "/home/julien/.local/lib/python3.8/site-packages/inkcut/ui/workspace.py", line 46, in start
    self.workbench.get_plugin('inkcut.ui')
  File "/home/julien/.local/lib/python3.8/site-packages/enaml/workbench/workbench.py", line 151, in get_plugin
    plugin.start()
  File "/home/julien/.local/lib/python3.8/site-packages/inkcut/ui/plugin.py", line 90, in start
    self.load_plugins()
  File "/home/julien/.local/lib/python3.8/site-packages/inkcut/ui/plugin.py", line 115, in load_plugins
    for entry_point in importlib.metadata.entry_points(
TypeError: entry_points() got an unexpected keyword argument 'group'</pre>

frmdstryr, 2025-04-16:
I believe https://github.com/inkcut/inkcut/commit/849526e5c5e00fcc3b0a3bbe862adb13e5ecc4f0 should fix this, can you try installing from the current master branch?

Pneumarian, 2025-07-06:
I'm getting the same issue on Devuan using a fresh install of inkcut 2.1.6

```
new_workspace.start()
  File "/usr/local/lib/python3.9/dist-packages/inkcut/ui/workspace.py", line 46, in start
    self.workbench.get_plugin('inkcut.ui')
  File "/usr/local/lib/python3.9/dist-packages/enaml/workbench/workbench.py", line 151, in get_plugin
    plugin.start()
  File "/usr/local/lib/python3.9/dist-packages/inkcut/ui/plugin.py", line 90, in start
    self.load_plugins()
  File "/usr/local/lib/python3.9/dist-packages/inkcut/ui/plugin.py", line 115, in load_plugins
    for entry_point in importlib.metadata.entry_points(
TypeError: entry_points() got an unexpected keyword argument 'group' 
```

[crash.txt](https://github.com/user-attachments/files/21083623/crash.txt)
[inkcut.txt](https://github.com/user-attachments/files/21083622/inkcut.txt)