inkcut/inkcut #401: How to connect a Mimaki CG-60sr with Inkcut

fintelkind, 2025-05-04:
Hi,
I have recently bought a Mimaki CG-60sr and would like to use it with Inkcut. As I didn't get any further after my first attempts, I searched the world wide web for answers. I didn't really find anything there either. 

The plotter is listed as a “Printer” in the list of possible devices. This is also mentioned in the installation instructions.
codelv.com/projects/inkcut/docs/device-setup
“Start your printer configuration utility (e.g. system-config-printer), which at least when connecting via USB should detect the connected cutter. Proceed to add it with the 'Generic' 'Raw Queue' driver.”

Neither Gnome nor Cups with all extensions (hplip, gutenprint, foomatic, etc) shows it.
One thing I could find is:

https://wiki.archlinux.org/title/CUPS/Troubleshooting#USB_printers
Conflict with usblp

USB printers can be accessed using two methods: The usblp kernel module and libusb. The former is the classic way. It is simple: data is sent to the printer by writing it to a device file as a simple serial data stream. Reading the same device file allows bi-di access, at least for things like reading out ink levels, status, or printer capability information (PJL). It works very well for simple printers, but for multi-function devices (printer/scanner) it is not suitable and manufacturers like HP supply their own backends. Source: [here](https://lists.linuxfoundation.org/pipermail/printing-architecture/2012/002412.html)[[dead link](https://en.wikipedia.org/wiki/Wikipedia:Link_rot) 2024-11-05 ⓘ].
Warning: As of [cups](https://archlinux.org/packages/?name=cups) version 1.6.0, it should no longer be necessary to blacklist the usblp kernel module. If you find out this is the only way to fix a remaining issue please report this upstream to the CUPS bug tracker and maybe also get in contact with Till Kamppeter (Debian CUPS maintainer). See [upstream bug](https://github.com/apple/cups/issues/4128) for more info.

If you have problems getting your USB printer to work, you can try [blacklisting](https://wiki.archlinux.org/title/Blacklisting) the usblp [kernel module](https://wiki.archlinux.org/title/Kernel_module):

/etc/modprobe.d/blacklistusblp.conf

blacklist usblp

Custom kernel users may need to manually load the usbcore [kernel module](https://wiki.archlinux.org/title/Kernel_module) before proceeding.

Once the modules are installed, run:

# journalctl -f

Then plug in the printer and inspect the output to see if it was detected.

If you are using usblp, the output should indicate that the printer has been detected like so:

Feb 19 20:17:11 kernel: printer.c: usblp0: USB Bidirectional
printer dev 2 if 0 alt 0 proto 2 vid 0x04E8 pid 0x300E
Feb 19 20:17:11 kernel: usb.c: usblp driver claimed interface cfef3920
Feb 19 20:17:11 kernel: printer.c: v0.13: USB Printer Device Class driver

If you blacklisted usblp, you will see something like:

usb 3-2: new full speed USB device using uhci_hcd and address 3
usb 3-2: configuration #1 chosen from 1 choice

Here is my dmesg output:
[ 6423.436966] usb 6-2: USB disconnect, device number 3
[ 7302.401925] usb 6-2: new full-speed USB device number 4 using uhci_hcd
[ 7302.558009] usb 6-2: New USB device found, idVendor=0a50, idProduct=0001, bcdDevice= 1.00
[ 7302.558017] usb 6-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 7302.558021] usb 6-2: Product: CG-SR
[ 7302.558025] usb 6-2: Manufacturer: MIMAKI
[ 7347.188964] usb 6-2: USB disconnect, device number 4
[ 7360.145945] usb 6-2: new full-speed USB device number 5 using uhci_hcd
[ 7360.303007] usb 6-2: New USB device found, idVendor=0a50, idProduct=0001, bcdDevice= 1.00
[ 7360.303015] usb 6-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 7360.303019] usb 6-2: Product: CG-SR
[ 7360.303022] usb 6-2: Manufacturer: MIMAKI

So, it looks to me as if there is a conflict between usblp and libusb. Both are modules loaded by the kernel to identify hardware and bind it into the system, right.
Elsewhere I've seen that a device has to be given udev rules, so that it will be writable. Here a short text:
https://debianforum.de/forum/viewtopic.php?t=135450
To create a suitable interface (e.g. /dev/ttyACM0), not only must a kernel module be available, but the device must also be correctly assigned. 

So through lsusb I've createtd this .rule:

ACTION=="add", SUBSYSTEM=="usb", ATTR{idVendor}=="0a50", ATTR{idProduct}=="0001", MODE:="664", GROUP:="lp", GROUP:="dialout

But nothing shows up in cups.

As I understand it:
1. when the plotter is plugged in, it must be mounted correctly.
That is, with the query demesg, the system should tell me that the device has been recognized as a plotter (printer) with the correct drivers and has been added to this path (/dev/usb*), right?
2. this will then be selectable as a local printer in Cups and added to the system.
3. is listed as a printer in Inkcut under Devices.

Anybody there who has some ideas how to proceed. Would be Grateful for any help.

fintelkind, 2025-05-17:
No one who could give me some advice??

frmdstryr, 2025-05-17:
> Feb 19 20:17:11 kernel: printer.c: v0.13: USB Printer Device Class driver

If the device is removed from the blacklist does it show up in inkcut the system as a printer?

frmdstryr, 2025-05-17:
From https://mimaki.com/archives/002/DB40257-05_CG-60SR_ENG.pdf, it looks like it uses a custom protocol Mimaki "MGL-IIc" whatever that is.

fintelkind, 2025-05-17:
> > Feb 19 20:17:11 kernel: printer.c: v0.13: USB Printer Device Class driver
> 
> If the device is removed from the blacklist does it show up in inkcut or the system as a printer?

Did you mean in /etc/modprobe.d?? Yea, I've tried to blacklist usblp in /etc/modprobe.d, but it didn't make a difference.

I've found this:
https://forums.debian.net/viewtopic.php?p=746927&sid=c8c249cb7bed5059ade133f3daaa4ead#p746927

There it says:
After more searching I found an article about this and the author thought Inkcut works with just about any cutter that uses HPGL commands, which includes most of the ones I've been looking at.

Most Manufacturers use the HP-GL (Hewlett-Packard Graphics Language) protocol for their Product and that's build around their specific driver for each Device, which you will find then in hplip or gutenprint repository, right??
So that means, there are no drivers for this protocol and therefore it will be not possible to run this plotter with linux.

This is why after journalctl-f it only says "found new high-speed usb device" but cound't bind it into the system, because otherwise it would direct to a path something like /usb0 or /tty0, right??

So, it looks like the only way to use this plotter will be with Windows.

fintelkind, 2025-05-17:
As Yian mentioned in the opening post at inkcut.org, he got a Mimaki CG-60st running and that has a similar protocol. 
https://inkcut.org/u/yian/296/

[image]

https://mimaki.com/archives/034/201301/CG60ST_Operation_D200417_V1.5.pdf

I'm bold enough to enter things into the Terminal and see it's outcome, but my knowledge what I'm actually doing there is minor. So I'm quiet lost here with this "situation", but my hope is still strong to get to something.

From my understanding, I have to get a device path for this plotter and from there onward I'm able to point at it from Inkcut or even through Inkscape.

fintelkind, 2025-05-18:
> > Feb 19 20:17:11 kernel: printer.c: v0.13: USB Printer Device Class driver
> 
> If the device is removed from the blacklist does it show up in inkcut or the system as a printer?

https://youtu.be/v-7yWtBRYLM
In the device specific properties shown at lsusb Robert is explaining in this video that if it doesn't comes up as Printer, then the system doesn't know where to put it.

[image]
[image]

And how is it with udev rules. Isn't it possible to give a specific device a system path and then use it as raw printer??

frmdstryr, 2025-05-18:
Sorry I'm confused what is the output of your device vs what is from other sources.

When I plug in my Roland Stika it shows the following and the desktop (KDE plasma) shows a popup saying there is no driver.

The `dmesg` output shows it is using builtin linux usblp driver.
```
[191117.200114] usb 3-5.1: new full-speed USB device number 9 using xhci_hcd
[191117.337679] usb 3-5.1: New USB device found, idVendor=0b75, idProduct=0174, bcdDevice= 0.00
[191117.337706] usb 3-5.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[191117.337715] usb 3-5.1: Product: SV-15
[191117.337722] usb 3-5.1: Manufacturer: Roland DG
[191117.337728] usb 3-5.1: SerialNumber: 0001
[191117.374241] usblp 3-5.1:1.0: usblp1: USB Bidirectional printer dev 9 if 0 alt 0 proto 2 vid 0x0B75 pid 0x0174
[191117.374267] usbcore: registered new interface driver usblp
[191122.466086] usblp1: removed
[191122.467168] usblp 3-5.1:1.0: usblp1: USB Bidirectional printer dev 9 if 0 alt 0 proto 2 vid 0x0B75 pid 0x0174

```

Going to system settings and clicking add a printer then choose "Raw" and "Raw Queue" then Save and Add. Then it should show up in Inkcut under the printers list (after refreshing if inkcut was already opened).

[image]

fintelkind, 2025-05-18:
here is my `dmesg` output:
```
[  251.487976] usb 6-2: new full-speed USB device number 2 using uhci_hcd
[  251.642995] usb 6-2: New USB device found, idVendor=0a50, idProduct=0001, bcdDevice= 1.00
[  251.643007] usb 6-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  251.643011] usb 6-2: Product: CG-SR
[  251.643015] usb 6-2: Manufacturer: MIMAKI
freshman@Precision-M4400:~$ 
```
nothing more.
And my thoughts where, that because of the missing printer information given by `lsusb` stated at "Interface Descriptor" that the actual "InterfaceClass" is _Vendor Specific Class_ and not _Printer_. So that's why the system doesn't know where to put it.

Actually the upper picture is mine (freshman@precision4400) and the lower one is the screenshot from the video.

frmdstryr, 2025-05-18:
I see. Can you try installing [pyusb](https://github.com/pyusb/pyusb/tree/master?tab=readme-ov-file#installing) in inkcut's python environment and see if you can access and write to the out endpoint as shown in this example https://github.com/pyusb/pyusb/blob/master/docs/tutorial.rst#lets-get-it-started from the "console" tab in inkcut?

You'll need to change idVendor in the line `dev = usb.core.find(idVendor=0xfffe, idProduct=0x0001)` to match your device (so use `idVendor=0x0a50`).

Then if it can access the endpoint try writing something like `ep.write('IN;PU0,0;PU100,0;')`

karliss, 2025-05-18:
>And my thoughts where, that because of the missing printer information given by lsusb stated at "Interface Descriptor" that the actual "InterfaceClass" is Vendor Specific Class and not Printer. So that's why the system doesn't know where to put it.

Yes that's one of main reasons. But it isn't as simple as system not knowing where to put it. Even configuring as "raw printer" usually requires usb device to implement common printer class interface it doesn't necessarily allow you to interact with raw usb device. The "raw printer" mostly just means that it printer doesn't use one of standard PDL(Page Description Language something like postscript, HP PCL) that the printer stack  would normally interact with and that it's the responsiblity of application to know all the details and printer stack just forwards it.

While the bottom layer of USB protocol for many devices might be common (low level USB commands),  and  top layer in many cases as well (everything is a file on linux), the middle layer how the low level usb commands map to a file like objects, on top of which the higher level protocols like HPGL or something else can be sent can be very different.

Usb is not just single simple protocol, it's also large collection of additional standardized protocols (more or less customizable)  for common device types which can be either directly supported by generic OS drivers or vender specific  drivers which build on top of generic or lower level USB drivers provided by OS.

For example removable block storage, usb to serial port adapter, printer, usb to parallel port might in the end all be exposed to userspace as file like objects. But the USB spec not only defines unique device class for each of them, but also what kind of commands they need to support and how they map to low level USB commands.  It's the responsibility of corresponding device driver to translate between low level USB commands and higher level OS interface for corresponding device type. You can't just take a USB flash drive tell the OS to use a usb/parallel port driver and have it work the exact format of commands differ.

In case of devices with custom device class you would normally have the device specific driver instead of generic device class one. The format of  commands expected by device might be as simple as directly writing your character  stream  containing in this case MGL-IIc2 into output endpoint split into max packet size chunks,  or could be something more complicated like custom packet structure of `size, command_id, command_payload, checksum` with the character stream being command_payload for specific command_id and various configuration and initialization commands have special command_id which go outside the character stream and MGL-IIc2 commands.

If you have a working windows setup it can be possible to reverse engineer the usb command structure by recording the USB traffic with wireshark. That's slightly more reliable process compared to blindly guessing and hoping that directly sending high level protocol to the usb endpoint will just work.

One more option is to reverse engineer the driver binaries, but that requires even more specialized skills. It's more complicated but at least you don't need a running setup.

fintelkind, 2025-05-20:
Thank you frmdstryr and karliss for your thought's. Now I got a picture of this situation and I must say for now I'm out. I have a unused Tower here and I will see that I get Windows running on that machine.
I should have looked a bit more into the details, then I would have chosen a different model like a Roland camm with a serial port, because a usb to serial would have been after all an option.
I'll see how it works with Wiresharks, but I'd say that's a few levels higher than my knowledge.
Will come back here if I get something interesting figured out.