inkcut/inkcut #405: Roland PNC-950 has wrong driver

vmario89, 2025-09-01:
Hi,

at the moment i am trying to revive our old PMC-950 to run on Linux instead Windows virtual machine (with Roland Cut Studio and proprietary drivers). It did not run for a very long time due to movement of our fablab to a new site and so on and so far.

i recognized that after installing Inkcut and configuring it Inkcut sends the job to the plotter but it fails. The plotter makes blinky-blinky when data is sent and that's it. It does not matter which dip switch settings for serial port or if i use parallel port. The permissions for usb-to-serial /usb-to-parallel are fine.

I was able to trace it down to the point that the PNC-950 does **not** use CAMM-GL1. Instead it used CAMM-GL3 which is also mentioned in the documentation PDF from the vendor:

[image]

the title says "CAMM-1" but its "CAMM-GL III". It's very weird to name it. But that's maybe the final reason.

We need some rewrite to support a  CAMM-3 protocol interface. I will try to compare CAMM-GL1 PRN files with CAMM-GL3 PRN files to see the differences and/or if i can send stuff manually to the plotter without Inkcut.

I tested our plotter with PNC-900 and PNC-1000 but it does not work too. Some other people reported it the past that this works. But not for me. I tried one usb-to-parallel adapter and two usb-to-serial adapers. With and without usb hub in the middle.

see also 

- https://github.com/inkcut/inkcut/issues/112
- https://github.com/inkcut/inkcut/pull/127 ( i dont know why i did this, because it does work anymore)
- https://github.com/inkcut/inkcut/pull/209/commits/061839e0b45684676e7869c9f5f2e92a82530817
- https://github.com/inkcut/inkcut/issues/212

vmario89, 2025-09-01:
projects regarding to CAMM-GL III i've found so far:
- `git clone https://2xor.de/git/dxfutils.git`
- https://github.com/fablab-neckar-alb/camm-gl3

for CAMM-GL I:
- https://github.com/MattTavares/senderCAMM-3

frmdstryr, 2025-09-01:
According to the `CAMM-Gl II -> programmers manual` above it says

> If no terminator is sent, the CAMM-2 may fail to operate

Can you try adding a `\r\n` after each write call in https://github.com/inkcut/inkcut/blob/master/inkcut/device/protocols/camm.py and see if it makes any difference?