inkcut/inkcut #406: Maximize window on application startup

vmario89, 2025-09-04:
No matter where i place maximize() function in the code, the app is never maximized (at least using Wayland). I want to start Inkcut in maximized window always, or to restore window size from previous app startup. I was able to adjust the x-y size in enaml manifest, but there is no easy function to get x y dims from the main screen properly. it also cannot be done with external tools like xdotool, ydotool or wlrctl :-/

any idea?

vmario89, 2025-09-10:
hey, thank you! i tried out but it does not work; or at least not with Wayland. Maybe its different with X11. Not sure

vmario89, 2025-09-10:
i think theres some bug in the library: it does something: when adding maximize, the window looks like this and has no shadow around:

[image]

when clicking maximize, it does NOT maximize, but adds shadow. Clicking maximize again, it gets finally maximized

[image]

jesbt, 2025-09-10:
I see, maybe it's a Plasma bug. I forgot to mention that I use Gnome.